hdl4fpga/hdl4fpga #5: Commit messages would be useful

emard, 2018-07-01:
I find this project is great and like to sometimes follow what happens.
I'd like sometimes to have a short 1-line comment to each commit message
something like "adjusted video colors" or "fix or mii to work with xyz".
Currently I view diff's , because the code is obvious :)

hdl4fpga, 2018-07-02:
Sorry I use github as place to store the work and download it to different computers.
I added DHCP and ARP to the core. Now you are not going to need to configure the LAN interface y you have DHCP. Also I am reworking the code so it will be easy for others to add things. I still remember what you asked me and I will try to do that as soon as possible.
Send me an email if you want to know any details

hdl4fpga, 2018-07-02:
The code that should be working is the released one. You can find it here https://github.com/hdl4fpga/hdl4fpga/releases/tag/arty. There, it is the code for lattice too. If that code doesn't work for you I will try to fix it. Is it the code that you are trying now ?

emard, 2018-07-02:
No that from the stable release

I was trying to fresh sync the project yesterday just from
curiosity and starting from ECP3 support, to port to ECP5 for
my ULX3S board, eventually add conversion from VGA to
onboard HDMI. I was thinking recent git has something that
works and replaces old stable release....

hdl4fpga, 2018-07-02:
Use the that release. The one that is on the repository is not working yet. Sorry about that

closed 2018-07-03 by hdl4fpga