hdl4fpga/hdl4fpga #8: ulx3s porting

emard, 2019-05-09:
HI

I'm trying to port your scope to ULX3S board, Goran sent me 
your message to fix constraints and toplevel so I did that and
even more

I have prepared toplevel and constraints to build and I
got my video test picture generator to display 800x600 picture, but
I can't compile your scope with option for 800x600
(that should be vlayout_id => 1) can you please fix your source for this
resolution?
Because for 1920x1080 I can't yet get test picture to work
(on other example it worked, but not now)

I have changed too much in order to compile
(replaced hdl4fpga -> work) so I don't pull request,
here is my attempt, see if you can take what is good and integrate
to your source

https://github.com/emard/hdl4fpga

hdl4fpga, 2019-05-09:
I will check it tomorrow .... bed time here ;-)

emard, 2019-05-09:
great. my actual project is in "diamond-cli" directory.

I couldn't compile from your diamond gui project, it gave me some
"-gu-gui" option error which I  don't know what to do about.

so I copied my cli scprits for compiling you can use makefile linux cli
or script-generated "ulx3s_85f_dvi.ldf". This project also works from
diamond gui

hdl4fpga, 2019-05-09:
I cloned your hdl4fpga and I got this

 ~/work/enard/hdl4fpga-master/ULX3S/diamond-cli $ make
xsltproc \
  --stringparam FPGA_DEVICE LFE5U-85F-6BG381C \
  --stringparam CONSTRAINTS_FILE ./constraints/ulx3s_v20.lpf \
  --stringparam STRATEGY_FILE ./scripts/ulx3s.sty \
  --stringparam XCF_FILE ./scripts/ulx3s_sram.xcf \
  --stringparam TOP_MODULE ulx3s \
  --stringparam TOP_MODULE_FILE ../common/ulx3s.vhd \
  --stringparam VHDL_FILES "../common/ulx3s.vhd ../common/clk_verilog.vhd ../common/hdl/vga.vhd ../common/hdl/vga2dvid.vhd ../common/hdl/tmds_encoder.vhd ../scopeio/scopeio.vhd ../../library/common/std.vhd ../../library/common/align.vhd ../../library/common/dpram.vhd ../../library/common/pipe_le.vhd ../../library/common/rom.vhd ../../library/common/bcddiv2e.vhd ../../library/common/vector.vhd ../../library/common/dbdbbl.vhd ../../library/common/ser2pll.vhd ../../library/common/stof.vhd ../../library/common/dtos.vhd ../../library/common/btod.vhd ../../library/common/btof.vhd ../../library/common/pll2ser.vhd ../../library/scope/scopeio.vhd ../../library/scope/scopeio_amp.vhd ../../library/scope/scopeio_downsampler.vhd ../../library/scope/scopeio_grid.vhd ../../library/scope/scopeio_miiudp.vhd ../../library/scope/scopeio_palette.vhd ../../library/scope/scopeio_segment.vhd ../../library/scope/scopeio_sin.vhd ../../library/scope/scopeio_tracer.vhd ../../library/scope/scopeio_trigger.vhd ../../library/scope/scopeio_axis.vhd ../../library/scope/scopeio_rgtr.vhd ../../library/scope/scopeio_ticks.vhd ../../library/scope/scopeio_formatu.vhd ../../library/scope/scopeio_istream.vhd ../../library/mii/miirx_pre.vhd ../../library/mii/mii_1chksum.vhd ../../library/mii/mii_cat.vhd ../../library/mii/mii_cmp.vhd ../../library/mii/mii_crc32.vhd ../../library/mii/mii_ipcfg.vhd ../../library/mii/mii_pll2ser.vhd ../../library/mii/mii_pllcmp.vhd ../../library/mii/mii_ram.vhd ../../library/mii/mii_rom.vhd ../../library/mii/mii_romcmp.vhd ../../library/mii/miitx_dll.vhd ../../library/video/video.vhd ../../library/video/video_win.vhd ../../library/video/cgafonts.vhd ../../library/video/cga_rom.vhd " \
  --stringparam VERILOG_FILES "../common/clk_verilog_v.v " \
  --stringparam SBX_FILES "" \
  ./scripts/ldf.xsl ./scripts/project.ldf > ulx3s_85f_dvi.ldf
LANG=C LD_LIBRARY_PATH=/mt/scratch/tmp/openfpga/prjtrellis/libtrellis /mt/scratch/tmp/openfpga/prjtrellis/libtrellis/ecppll --module=clk_verilog_v --clkin=25 --clkout0=200 --clkout1=40 --clkout2=40 --clkout3=40 --file ../common/clk_verilog_v.v
/bin/sh: /mt/scratch/tmp/openfpga/prjtrellis/libtrellis/ecppll: No such file or directory
make: *** [scripts/diamond_main.mk:165: ../common/clk_verilog_v.v] Error 127

Where can I get the missing files ?

emard, 2019-05-09:
HI the PLL clock generator I'm using from my own fork of
prjtrellis PLL generator for ECP5 open source tools, source is here
https://github.com/emard/prjtrellis/blob/master/libtrellis/tools/ecppll.cpp
prjtrellis needs to be compiled to get ecppll exe

but you already have this clock just disable clock generation
options in makefile and it will use the one that's already there
that generates 200 MHz and 40 MHz for 800x600

# git stash
# disable automatic generation of the clock
CLK0_NAME = clk_verilog_v_disabled
# use explicit file that's already in my git
VERILOG_FILES = ../common/clk_verilog_v.v

For me it's faster to make PLL from cmdline but you can normally use
diamond GUI to generate the clock,
for 800x600 it's 40 MHz pixel clk and shift clk 40*5 = 200 MHz
similar for 1024x768  65 MHz, 325 MHz, 1920x1080 75MHz, 375 MHz

emard, 2019-05-09:
also, If you just load "ulx3s_85f_dvi.ldf" from diamond gui,
it should completely compile without rebuilding this clock pll

hdl4fpga, 2019-05-09:
I got this error now ..

hdl4fpga-master/ULX3S/diamond-cli $ make
xsltproc \
  --stringparam FPGA_DEVICE LFE5U-85F-6BG381C \
  --stringparam CONSTRAINTS_FILE ./constraints/ulx3s_v20.lpf \
  --stringparam STRATEGY_FILE ./scripts/ulx3s.sty \
  --stringparam XCF_FILE ./scripts/ulx3s_sram.xcf \
  --stringparam TOP_MODULE ulx3s \
  --stringparam TOP_MODULE_FILE ../common/ulx3s.vhd \
  --stringparam VHDL_FILES "../common/ulx3s.vhd ../common/clk_verilog.vhd ../common/hdl/vga.vhd ../common/hdl/vga2dvid.vhd ../common/hdl/tmds_encoder.vhd ../scopeio/scopeio.vhd ../../library/common/std.vhd ../../library/common/align.vhd ../../library/common/dpram.vhd ../../library/common/pipe_le.vhd ../../library/common/rom.vhd ../../library/common/bcddiv2e.vhd ../../library/common/vector.vhd ../../library/common/dbdbbl.vhd ../../library/common/ser2pll.vhd ../../library/common/stof.vhd ../../library/common/dtos.vhd ../../library/common/btod.vhd ../../library/common/btof.vhd ../../library/common/pll2ser.vhd ../../library/scope/scopeio.vhd ../../library/scope/scopeio_amp.vhd ../../library/scope/scopeio_downsampler.vhd ../../library/scope/scopeio_grid.vhd ../../library/scope/scopeio_miiudp.vhd ../../library/scope/scopeio_palette.vhd ../../library/scope/scopeio_segment.vhd ../../library/scope/scopeio_sin.vhd ../../library/scope/scopeio_tracer.vhd ../../library/scope/scopeio_trigger.vhd ../../library/scope/scopeio_axis.vhd ../../library/scope/scopeio_rgtr.vhd ../../library/scope/scopeio_ticks.vhd ../../library/scope/scopeio_formatu.vhd ../../library/scope/scopeio_istream.vhd ../../library/mii/miirx_pre.vhd ../../library/mii/mii_1chksum.vhd ../../library/mii/mii_cat.vhd ../../library/mii/mii_cmp.vhd ../../library/mii/mii_crc32.vhd ../../library/mii/mii_ipcfg.vhd ../../library/mii/mii_pll2ser.vhd ../../library/mii/mii_pllcmp.vhd ../../library/mii/mii_ram.vhd ../../library/mii/mii_rom.vhd ../../library/mii/mii_romcmp.vhd ../../library/mii/miitx_dll.vhd ../../library/video/video.vhd ../../library/video/video_win.vhd ../../library/video/cgafonts.vhd ../../library/video/cga_rom.vhd " \
  --stringparam VERILOG_FILES "../common/clk_verilog.vhd" \
  --stringparam SBX_FILES "" \
  ./scripts/ldf.xsl ./scripts/project.ldf > ulx3s_85f_dvi.ldf
echo prj_project open ulx3s_85f_dvi.ldf \; prj_run Export -task Bitgen | 
/bin/sh: -c: line 1: syntax error: unexpected end of file
make: *** [scripts/diamond_main.mk:159: project/project_project.bit] Error 1

emard, 2019-05-09:
I will change it not to rebuild the clock, test if it compiles for me then
give me few minutes

emard, 2019-05-09:
HI

I commited to the git the change in makefile to avoid calling
ecppll and trellis tools completely (I hope)

so make clean; make should generate bitstream
yes you need to set diamond tools correct path in file
scripts/diamond_path.mk
and also all "sh" need in-place to be edited (changed) for
"bash" in complete diamond installation, that is common
problem if using debian/ubuntu
to fix that, I have automatized script here

https://github.com/emard/ulx3s-bin/tree/master/fpga/diamond/diamond-fix-scripts/change-sh-to-bash

hdl4fpga, 2019-05-09:
Ok ... It seems it works now ...
I got this error that has to do with programming tool ....
Loading Programmer Device Database...

Generating VME.....
Reading Input File: ulx3s_85f.xcf
ERROR - Device1 LFE5U-85F: File ulx3s_85f_dvi.bit is invalid.

Check configuration setup: Unsuccessful.

ERROR - File ulx3s_85f.xcf is invalid.
make: *** [scripts/diamond_main.mk:190: ulx3s_85f_dvi.vme] Error 249

hdl4fpga, 2019-05-09:
So the problem was the resolution .... have you change the video_clk frequency ?

hdl4fpga, 2019-05-09:
Forget it ... I haven't read the code already.

hdl4fpga, 2019-05-09:
I see that you still have 	vlayout_id  to 0 on your code ... that is for 1080p
https://github.com/emard/hdl4fpga/blob/master/ULX3S/scopeio/scopeio.vhd

	scopeio_e : entity work.scopeio
	generic map (
		vlayout_id  => 0 -- 0:1920x1080, 1:800x600

You should set it 1 but maybe it still doesn't work on scopeio as I was reworking the code. But I think that you should see something  setting vlayout_id to 1 ... I think this weekend It will work on 800x600

hdl4fpga, 2019-05-09:
You can set so_clk, si_clk, and input_clk to '0' with its corresponding data signals too,  you get an image on the screen.

scopeio_e : entity work.scopeio
	generic map (
		vlayout_id  => 1 -- 0:1920x1080, 1:800x600
	)
	port map (

		si_clk      => '0´,
		si_frm      => '0',
		si_data     => (0 to 0 => '0'),
		so_clk      => '0',
		ipcfg_req   =>  '0',
		input_clk   => '0',
.....

emard, 2019-05-09:
When I set vlayout_id  => 1 then it won't compile, here's error:

@E: CD297 :"/tmp/hdl4fpga/library/scope/scopeio.vhd":895:6:895:15|Width
mismatch, location has width 13, value 20

hdl4fpga, 2019-05-09:
Ok .... that was that I was referring to when I said  I am reworking the code ... I will try to solve it for tomorrow. sorry about that

emard, 2019-05-09:
Yes that's ok!
In the meantime can you also make sure that vga_blank
is generated? HDMI output needs blank signal to show
picture, while VGA monitors don't need blank

hdl4fpga, 2019-05-09:
Isn't it working ? I thought it was ... I've never touched it ... I will check. I used it because  sometimes vga LCD panel needs RGB to be 0 to sync.

emard, 2019-05-09:
I can't know does it work yet, but missing or wrong blank signal
is common source of the HDMI no-sync problems :)

we will know better when you fix 800x600 compile :)

In the meantime, I can only try to fix my code for 1920x1080 to show picture.
I don't know why it doesn't work for me now, maybe monitor has changed.
1280x1024 works, it has the same 75 MHz pixel clock and 1920x1080 should work
but doesn't sync :(...

emard, 2019-05-10:
HI

I got my vga test picture on 1920x1080 @ 30Hz on another monitor with
this settings:

--      -- 1920x1080 75 MHz pixel clock, works on samsung TV
      C_resolution_x => 1920,
      C_hsync_front_porch => 88,
      C_hsync_pulse => 44,
      C_hsync_back_porch => 133,
      C_resolution_y => 1080,
      C_vsync_front_porch => 4,
      C_vsync_pulse => 5,
      C_vsync_back_porch => 46,

and then I tried your code to generate picture so I added mode 9
constant h_tab : natural_matrix (0 to 9, 3 downto 0) := (
                9 => (1920,  88,  44, 133)); -- 1920x1080R@30Hz pclk  75.00MHz
constant v_tab : natural_matrix (0 to 9, 3 downto 0) := (
                9 => (1080, 4, 5, 46)); -- 1920x1080R@30Hz pclk  75.00MHz
constant vlayout_tab : vlayout_vector(0 to 1) := (
                0 => (    9,       1920,           4,          50,
       8,          8,       6*8,         33*8,       1,        0,
 1),

But then I get no picture. I'm not sure of meaning of your timing values,
can you add some comment in the code like this:
(resoluton, front_porch, pulse, back_porch)

hdl4fpga, 2019-05-10:
I found the mistake on https://github.com/hdl4fpga/hdl4fpga/blob/master/library/scope/scopeio.vhd. It compiles now, but I didn't try it on a kit yet.

video_blank was  active-low. Now  it is active_high.

Here you can find video timing parameter.
https://github.com/hdl4fpga/hdl4fpga/blob/master/library/video/video.vhd
sequence is : active, front porch, sync, back porch.

800x600 
horizontal timing
         active, front port,    sync,     back port 
1 => ( 800,             32,       80,             112), --   800x600C@60Hz pclk  38.25MHz

vertical timing.
         active, front port,    sync,     back port 
1 => ( 600,              3,          4,               17),	--   800x600C@60Hz pclk  38.25MHz

hdl4fpga, 2019-05-10:
It seems to be working on 800x600 now.

emard, 2019-05-10:
800x600 IS WORKING on ULX3S now!

I just copied your library/scope/scopeio.vhd to my fork
and picture appeared with some dumped sinewaves
on 2-channel scope and blue cyan and gray borders around
the grids...

Wonderful!

emard, 2019-05-10:
I can also report that 1920x1080 @ 30Hz at 75 MHz pixel clock is
now showing correct picture on my Samsung TV

this mode is done using timings from mode 9 I posted at few
messages above (or at my fork library/video/video.vhd) so
you can add this mode to your sources.

At ULX3S it's amazing how small it is - for 4 channels 1920x1080 only
1k LUT and 7 BRAMs are used, so there's is going to be a plenty of
space left for additional features.

hdl4fpga, 2019-05-10:
Great Job. Nice team work. It took only one day and a half to port the code to the ULX3S kit. Neither did you get a documentation nor a commented code, and neither did I get the kit. Also you added the a new video interface.

Thanks ... that was something that I really wanted to know: how hard it would be for somebody but not I to port scopeio to another kit.

I added the new video sync mode 9 and I will add the new layout description.

It is supposed the signal be a sync, and one segment  follows the other in time. Colors are configurable ... nobody agrees what are the nicest XD

As I told to goran.mahovlic, when a new issue is opened I don't get an email from github. I willl leave this issue open till the port is finished ... but if you open a new issue and I don't reply in a day, you can contact me on https://discord.gg/qMBGW6A

So now ... we have to deal with controller interface XD

hdl4fpga, 2019-05-10:
Can you send me a photo ?

emard, 2019-05-10:
documentation? no thanks, as long as I can understand the code :)

Video interface just copypasted from my ulx3s-misc example project,
Also I'm maintaining f32c project for multiplatform
(xilinx/altera/lattice, 20-30 boards supported)
so I have portability experience and scopeio was for me easy.

It would be intermediately hard (possible in a week) for
user who knows VHDL 1-2 years and has a kit with
similar/compatible FPGA and onboard hardware.

It would be difficult or impossible for novice user who just
started learning VHDL or who has completely different
FPGA chip or very different hardware than those already supported.

They should normally send you the board for free in order to
get ported.

For controller interface, I plan to add PS/2 mouse driver core
and make a simple example for you to see how the mouse pointer
can move over the scope screen.

Some simple 1-bram lookup table can be used as simple search tool
for rectangular regions and to detect simplle type of user action
(click or drag)
which will change appropriate parameter. No CPU, no X11 just simplle logic.

emard, 2019-05-10:
If you need a photo of ULX3S running scope on monitor,
I'm not with camera here now Goran will publish pictures of
this tonight I guess.

hdl4fpga, 2019-05-10:
Thanks ... I don't need the photo but I like to see it

emard, 2019-05-10:
HI I added PS/2 mouse support here
Mouse moves crosshair pointer over the scope screen
nothing to be clicked yet but you get idea what should be done.

https://github.com/emard/hdl4fpga/commits/master

mouse crosshair is drawn a "hacky"  way in scopeio.vhd,
because I don't completely understand your pallete logic,

Can you rewrite it correctly to follow your programming logic?

hdl4fpga, 2019-05-10:
Yes, I will make room for a mouse maybe tomorrow ... XilinX ISE is killing me with another one bug

emard, 2019-05-10:
Spartan 3E has nice PS2 mouse port :)

When I have problems with ISE I try few different
command line parameters or have to rewrite the code.
Usually it's possible, I know how difficult is to write
portable code that works the same on different FPGA

I wil continue to experiment with mouse on
1920x1080 so far this "technology" seems promisiing :)

hdl4fpga, 2019-05-10:
Fix it ... thanks ... synplify(lattice) is so good compare to ISE ... I have to rewrite several vhdl code so ISE doesn't crash on Linux and WIndows also Vivado .... XD

emard, 2019-05-10:
Tell me about it :) after a lot of code changes I have automatized myself
to write portable VHDL even when I don't have to port anywhere

Diamond compiles fast, it's not to big on disk,
it infers well and bistreams perform well, and it
doesn't crash too often. I'm using version 3.7
and won't upgrade soo because 3.7 was the best

emard, 2019-05-10:
HI

I moved mouse and dvi video to hdl4library and
updated diamond ldf project generator scripts to
support library hdl4fpga so your sources now compile
unmodified (no need to replace "hdl4fpga." -> "work." anymore)

you can also load such generated lattice diamond project from
GUI and it will compile too

you can merge at this point if you agree how is done now

hdl4fpga, 2019-05-10:
I mixed up every thing ... I wanted to commit ULX3S only 
I will check it tomorrow

emard, 2019-05-10:
I will rewrite mouse core to VHDL to avoid verilog in your library.

hdl4fpga, 2019-05-10:
No ... keep it that way ... I will tell it tomorrow ... please give me sometime ;) I have to finish something else

emard, 2019-05-10:
I have now both versions, vhdl and verilog, both work the same
you can choose whatever you like.

never mind conversion, oberon mouse was originally written in
lola-2 anyway :)

emard, 2019-05-12:
HI

At my latest commit, I got serial port working at ULX3S.
it can change colors and scope parameter, I tested it by
stty 115200 < /dev/ttyUSB0
cat < /dev/urandom > /dev/ttyUSB0

colors change, numbers change, and time/scale changes
randomly... so far so good :)

but from firefox debug.html opens some sliders, I can move them
but serial port seems not to be working
(security, settining, should I setup firefox somehow for this?

hdl4fpga, 2019-05-12:
Great! ... I didn't know that firefox could handle serial port  THAT's GREAT!. Serialport and NW.js are a little bit annoying 

You write code faster than I can write mine. lol ... Still writing code for UART!

I hope to start writing code for mouse on Monday.

emard, 2019-05-12:
You may need to try several PS/2 or USB+PS/2 mouse's
to find out the one which is working. Or ask me for a clue
to fix

Very simple initialization sequence is performed and also
I'm feeding mouse core with too fast clock, it's now
pixel clock (75MHz and 40MHz) but mouse core is written
to for 25 MHz.

My all logitech and microsoft mouses are working with this,
xy direction, wheel rotation and 3 click buttons have signals
so you should have plenty options to construct cool GUI

hdl4fpga, 2019-05-12:
Take I for granted that I'll ask you ...

hdl4fpga, 2019-05-12:
I couln't make firefox work the serial port ? Do you have a link to any doc ? I couldn't find any doc on Internet as I could find serialport for nw.js or Electron

emard, 2019-05-12:
HI

No no neither I can get firefox serial port working, see I wrote
this because I was actually asking for help too.
I didn't know what NW/electron is (never used that before)

...but from firefox debug.html opens some sliders, I can move them
but serial port seems not to be working...

So firefox opens GUI, but no data is sent to serial port from my linux/debian

So I closed firefox and tested serial port just by constantly dumping random
numbers to ttyUSB0 and every few seconds scopeio recognizes something
from random sequence as a "valid" packet and does some change...

hdl4fpga, 2019-05-12:
The nw.js page and serialport page will explain what they are better than I.
nw.js https://nwjs.io/
serialport https://serialport.io/

I added a script to install both of them automatically. you need to install npm first or nodejs. Both packages come with most Linux distributions.

you should be able to connect to the serial port 

Something happens with the trigger, it is not working on scopeio ... may be I've broken something

emard, 2019-05-12:
I can ask Goran to try this NPM serial port,
it's more complicated than I thought.

For this I prefer something simple like a python
script to send few serial bytes over commandline...

I have seen trigger problems too,
It works good for repeatable signals like
sine or sawtooth wave but if
scoping PS/2 traffic it looks a bit chaotic,
sometimes it doesn't trigger, and when it
triggers, start of PS/2 is not on T=0 but somehow
random on time graph.

emard, 2019-05-12:
Can you document few simple serial demo commands like e.g.

to set trigger rising edge with something like:
echo -ne "\x5C\xAA\xBB\xCC\x00" > /dev/ttyUSB0

hdl4fpga, 2019-05-12:
I hope this is useful https://github.com/hdl4fpga/hdl4fpga/wiki. I will complete later but tell me if it is clear enough

emard, 2019-05-13:
Yes but I think doesn't work for me.
when I do this from above examples,
nothing changes:

stty raw 115200 < /dev/ttyUSB0
echo -ne "\x10\x02\x12\x12\x32\x00" > /dev/ttyUSB0
echo -ne "\x11\x5C\x00\x61\x00" > /dev/ttyUSB0

at ULX3S I have OLED hex decoder display so I tested
sending each byte manually. OLED display
shows correct each value, so I'm shure it's received
by UART but no visible reaction of scopeio display.

but this changes something:
cat /dev/urandom  > /dev/ttyUSB0

emard, 2019-05-13:
Hey!

After running urandom, I stopped it with ctrl-c and
then, this command worked one time:
echo -ne "\x10\x02\x12\x12\x32\x00" > /dev/ttyUSB0
I saw yellow trace moved down.

but it worked only once, then i changed to
echo -ne "\x10\x02\x12\x22\x12\x00" > /dev/ttyUSB0
nothing changed, I tried urandom trick again but no effect.

Do you think something is missing in my echo sequence?

hdl4fpga, 2019-05-13:
everything seems to be fine ... i would try sending 0x00 0x00 at the
begining to asure an end of packet before sending a new packet ... but your
sequence is correct ... ill check it tomorrow

emard, 2019-05-13:
I tried this i a loop, multiple times but it doesn't work:

stty raw 115200 < /dev/ttyUSB0
/bin/echo -ne "\x00\x00\x10\x02\x12\x12\x32\x00" > /dev/ttyUSB0

hdl4fpga, 2019-05-13:
Can you try this loop ?

begin loop
0x00 0x00 0x11 0x5c 0x00 0x01 0x00       // set grid bg black
0x00 0x00 0x11 0x6c 0x00 0x61 0x00       // set grid bg yellow
end loop

you should see the grid bg  flashing yellow

hdl4fpga, 2019-05-13:
maybe 0x00 is interpreting as end of string by echo and it stop there. as far as i remember argv does not  have the length of the argument .. it uses a null 0x00 as end of the string

emard, 2019-05-13:
/bin/echo -ne will send \x00 as NUL-char and continue
I tested every char individualy and I see it's hex code on
OLED HEX display

but this doesn't work, nothing visible happens to the grid
or video at all:
btw you mistyped 5c escape seq with 6c so I replaced back 5c

#!/bin/sh
stty raw 115200 < /dev/ttyUSB0
while true
do
  # // set grid bg black
  /bin/echo -ne "\x00\x00\x11\x5c\x00\x01\x00" > /dev/ttyUSB0
  sleep 1
  # set grid bg yellow
  /bin/echo -ne "\x00\x00\x11\x5c\x00\x61\x00" > /dev/ttyUSB0
  sleep 1
done

emard, 2019-05-13:
test of echo nul-char capability:

/bin/echo -ne "\x00\x00\x11\x5c\x00\x61\x00" | hexdump -C
00000000  00 00 11 5c 00 61 00

hdl4fpga, 2019-05-13:
I don't get it ... there was a bug that I t solved few days ago. It was in https://github.com/hdl4fpga/hdl4fpga/blob/master/library/scope/scopeio_istream.vhd, This file controls the stream escape sequence but you have it up to date on repository.

emard, 2019-05-13:
Do you think here we may have mixed clock domain issue?

hdl4fpga, 2019-05-13:
dont think so

emard, 2019-05-13:
At your board, is my echo sequence working?
I see also mixed bit order, things (0 to 7) stored to (7 downto 0)
that doesn't look very nice to me...

[345 more replies]

closed 2019-07-04 by hdl4fpga