hdl4fpga/hdl4fpga #97: trace rendering bug on the right side of grid

emard, 2019-07-03:
On 800x600, layout sd600x16fs
at right side of grid (at the last half of rightmost div)
the X-part of traces which actually belongs to the left side of grid
is shown on the right side, like some wraparound

on this right area when bug appears also the traces step up by 1 pixel

hdl4fpga, 2019-07-03:
Ok .. I will check it .. I see many strange things too

hdl4fpga, 2019-07-04:
It is fixed it. No wraparound, it was only a delay on box's y coordinate.

vtaxis_width and hzaxis_heiight can be made 0 independently. scales are moved inside the grid when those values are set, saving space on small display.

emard, 2019-07-04:
I tried it on ULX3S but there's some bug -
each segment is completely filled with segmentbg color.

I tried layouts sd600x16 sd600x16fs oled64x96
it's always the same

no traces, grids, numbers.

emard, 2019-07-04:
it synthesizes to 930 LUTs while before it was abut 1900 LUTs

hdl4fpga, 2019-07-04:
try it now

emard, 2019-07-04:
I changed scopeio.vhd to scopeio_xxx.vhd, then it synthesizes to
a real size of 1830 LUTs and grid numbers and traces shown
but colors control of hzscale and vtscale seems either changed
or not working

emard, 2019-07-04:
I'll try now the latest scopeio.vhd

hdl4fpga, 2019-07-04:
Some of palettes_ids have changed to make scales to be moved inside of the
grid.

emard, 2019-07-04:
latest scopeio.vhd still syntesizes to 927 LUTs

hdl4fpga, 2019-07-04:
I´ll check it ...I haven't change it much ...only to solve bugs

emard, 2019-07-04:
Yes something is like optimized out,
you can try to compile for my board just look for the
LUT size, if its 930 LUTs its too small, if its 1830 it's OK

hdl4fpga, 2019-07-04:
I've found it ...! I forgot one line

emard, 2019-07-04:
just commit it and I will try

hdl4fpga, 2019-07-04:
Ready    I hope

emard, 2019-07-04:
compiling

emard, 2019-07-04:
1783 LUTs so far soo good, waiting 4 bitstream ... :)

hdl4fpga, 2019-07-04:
Ok ... waiting

emard, 2019-07-04:
I got grid and traces on layout 1 standard sd600 (800x600)!

I need to fix mouse2rgtr a bit for changed palleteid
for the color of hzscale numbers on the bottom

emard, 2019-07-04:
The reported bugs seems really fixed :)

Can I also export axis_scale generic to scopeio.vhd
so every board can configure its own numeric scale

hdl4fpga, 2019-07-04:
Yes ... you can ... it is there for that .... but take into account that in
the future there could be some changes

emard, 2019-07-04:
I'll just expose it simple as is for now.

in the future you will change it of course to something
more like separate scale for X/Y etc

emard, 2019-07-04:
I think the trace wraparound bug is still here
on sd600x16fs

I will compile it for other layouts to see maybe
some have it fixed some not

emard, 2019-07-04:
layout sd600 has it fixed

hdl4fpga, 2019-07-05:
One sample may be ... I have to see how to caliber scopeio for one sample
... another thing TODO

hdl4fpga, 2019-07-05:
> layout sd600 has it fixed
>
>
> > I think the trace wraparound bug is still here
> > on sd600x16fs
I've seen  somthing wierd ...

hdl4fpga, 2019-07-05:
I've moved out of scopeio_xxx.vhd the video system. I put it on a file called scopeio_video..vhd.
Also, the resize system was moved out to scopeio_resize.vhd.
Now the samples don't overflow but saturate.
Can you take a look at it to make your code compatible with scopeio_xxx.vhd to make it the main scopeio.vhd ?

emard, 2019-07-05:
OK!

if I understand correctly, scopeio_xxx.vhd is now the candidate
to become scopeio.vhd and current scopeio.vhd moved to old/ or something

so I will now adapt scopeio_capture1shot.vhd to become
compatible with current scopeio_xxx.vhd

emard, 2019-07-05:
I got first problem compiling:
@E: CD297 :"/home/guest/src/fpga/hdl4fpga/library/scope/scopeio_segment.vhd":259:7:259:28|Width
mismatch, location has width 8, value 32

emard, 2019-07-05:
It compiles only for 1 channel so I changed to 1
(It had 4 channels)

emard, 2019-07-05:
my scopeio_capture1shot.vhd works with your current scopeio_xxx.vhd

        scopeio_capture_e : entity hdl4fpga.scopeio_capture1shot
-- it needs additional signals:
                video_vton    => video_vton,
                trigger_freeze=> trigger_freeze,
                trigger_shot  => trigger_shot,

bugs known are
1. only 1-channel bug
2. trace rendering bug on right side (this issue)
3. when frozen, traces don't horizontally scroll with hzscale
(I can probably try to fix this in capture1shot for now...)

hdl4fpga, 2019-07-06:
i've fixed the width mismatch ... i haven't tried it

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

emard, 2019-07-06:
great, also grid alignment in capture is correct now,
so -1 not needed

hdl4fpga, 2019-07-06:
Fixed

closed 2019-07-06 by hdl4fpga