hdl4fpga/hdl4fpga #110: scopeio_xxx 1-shot trigger

emard, 2019-07-05:
HI

I ported my triggering logic from scopeio.vhd to scopeio_capture1shot.vhd
Few examples seem to work same as scopeio.vhd except that horizontal
scrolling doesn't work if the trigger is frozen.

Perhaps the horizontal scrolling with "input_delay" doesn't belong
to capture module anyway, it should be moved to scopeio_xxx or elsewhere
(I moved horizontal scrolling to scopeio.vhd)

hdl4fpga, 2019-07-05:
It's not complete ... trigger is sketchy .... don't port anything to that till i tell you ... it is something that we have to coordinate very well to do it properly.

emard, 2019-07-05:
Of course! This is not a serious port,
just a proof of concept that some module like this still
can work with new scopeio_xxx.vhd.

Let me know when triggering system will be reworked,
I don' want  to miss the party .-.. :)

hdl4fpga, 2019-07-05:
I'm fixing the overflow.

emard, 2019-07-07:
on ulx3s I got compile error:
@E: CD455 :"/home/guest/src/fpga/hdl4fpga/library/common/std.vhd":548:3:548:8|Slice's bounds are outside the variable's bounds
                        retval(i*size to (i+1)*size-1) := to_unsigned(data(i), size);

hdl4fpga, 2019-07-07:
Fixed

emard, 2019-07-07:
Thanx :)

closed 2019-07-07 by emard