hdl4fpga/hdl4fpga #118: USB mouse written dirty

emard, 2019-07-06:
This is not real issue but a comment,

I took core from https://github.com/renaudhelias/CoreAmstrad
He made this also by scoping functional protocol between USB
joystick and PC and then writing the VHDL code..

and too bad it's written more likely as C than as VHDL.
From synthesis to synthesis it either works or doesn't
chances are about 50%. Just change anything unrelated
recompile and try again...

It's USB bus is now connected to scopeio so at least one
can have better insight in what is going on with signals
and timings

hdl4fpga, 2019-07-09:
I see you are using scopeio_xxx.vhd , Do you still need scopeio.vhd ? if you don't, I'll move scopeio_xxx.vhd to scopeio.vhd

emard, 2019-07-09:
Everything works on scopeio_xxx.vhd, I recently
used it to watch USB traffic, the waveforms
look great.

You can make it default, old scopeio.vhd is no
longer needed

emard, 2019-07-18:
usbmouse won't be fixed, but I will keep on searching over the
internet if I can find a better core which supports usb mouse
more reliably

closed 2019-07-18 by emard