hdl4fpga/hdl4fpga #127: trace rendering bug again but smaller

emard, 2019-07-07:
at layout sd600x16fs

digital squarewave, on 1st segment on the left
(the very first sample shown on screen), When i
scroll horizontally left-right, traces on some scroll
position make fake edge logic 0-1 transition.
Happens only at 1st segment, 1st sample on the left.
Seems that this fake "edge" is taken from another
part of the storage buffer

Its not a big bug but just a bit annoying,
showing edge where it shouldn't be one

hdl4fpga, 2019-07-08:
I'll try to reproduce ... I don't want any annoying thing ...  if you can send a picture it will be very helpful ... those bugs are very difficult to catch and very annoying to the eyes

emard, 2019-07-08:
HI!

I have pictured a waveform that demonstrates this bug when
scrolling a frozen waveform.
https://github.com/emard/hdl4fpga-bugs

hdl4fpga, 2019-07-08:
Can you edit with gimp or some like that and show me exactly what is wrong ? because if it is what I think it is not a bug. Also I need to see the right part too

emard, 2019-07-08:
HI I edited the bug picture and it is

IMG_0004_BUG_MARKED.JPG

the rest pictures show correct buffer, there is nothing else
in the buffer but this waveform which is displayed.

In the marked picture you will see edge that appears on
traces and this edge is non-existent in the signal waveform

hdl4fpga, 2019-07-08:
I need to see the right part ... because if it is what I think ... you are watching the rising and falling edge of signal that is a continuation of the right part of the previous segment. And those samples are just the exact moment when it is happening

emard, 2019-07-08:
both left and right part should be visible on jtag picture
is it that or something else right

hdl4fpga, 2019-07-08:
Ohh... sorry ...

hdl4fpga, 2019-07-08:
You told me that that happens when it is frozen, isn't it ?

hdl4fpga, 2019-07-08:
Also, if you scroll it ... are those sample keeping on the same place ?

emard, 2019-07-08:
It happens both in frozen and unfrozen picture if the signal
is triggered as stable on the same screen position.

This "artefact" edge is copy/alias of the vertical edge of the
waveform cca 48 samples from the left side, so artefacted
samplle 1  blinks together with real sample 48 if  the waveform
is actively triggered. This 48th sample is visually
approx  determined, it may be few samples +- of course.

artefact edge is not wraparound of anything from the right side of the
segment or the far end of the buffer. It appears only if the 48th sample
has edge.  If 48th sample is part of a flat line, the artefact edge is not
appearing

as I scroll waveform, artefact edge sample always appears as vertical
line at the 1st sample left from the segment. As I scroll horizontally, it
appears to me as it is alias of 48th line of the same segment.

It can appear at any/each segment which has edge on 48th sample

hdl4fpga, 2019-07-08:
I got it ... it is taking the last box x position of the scale to get the sample ... can you try this please ?
Compile it with layout's vtscale_width set to 40 and confirm if the sample is cca 40

emard, 2019-07-08:
yes, I compiled with
vtaxis_width     =>   40,
and the artefact line now copies cca 40th sample from the segment

hdl4fpga, 2019-07-08:
There are two places where the bug is ... first one is: ... there is an extra delay in scopeio_capture1shot, second one : it is in scopeio_video.vhd ... I'll check scopeio_video.vhd ... Can you check scopeio_capture1shot.vhd ?

emard, 2019-07-08:
What delay means related to video display actually?
There seems to me there's just standard BRAM 1-cycle read latency.

I would like to suggest that scrolling of captured content
should be moved out of capture_* module. My proposal would
be that even BRAM is moved out of capture1shot.
Capture1shot module will have only bram write bus interfaces
clk,addr,data,wr, bram and the BRAM and the rest of video scrolling
logic can move outside to scopeio_xxx.vhd or in separate module
scopeio_storage for example.

emard, 2019-07-08:
aiieee I see there is extra registered logic
around captured_addr arithmetic
I can make this combinatorial but we'd better really
move it outside of capture module....

hdl4fpga, 2019-07-08:
I have a signal with glitches and I wasn't able to reproduce the bug.
Here is mine https://github.com/hdl4fpga/hdl4fpga/blob/work/images/20190708_144608.jpg

hdl4fpga, 2019-07-08:
take a look at scopeio_capture.vhd ... that is what I am point at.
Trigger modes should be out of capture system.

emard, 2019-07-08:
hmm... but I think I can't LUT-efficiently make 1shot triggering separate of
the sample recording logic because both parts cooperate in using of the
counters and mutual resets to precisely control center of waveform around T0
and deflickering inside of the buffer. There is only one buffer, at
the same time
it must be written by the capture and read for display and look like something
visually stable waveform (e.g. sinewave)

If current 1shot trigger system is separated, whole logic will need to
be duplicated
and not loosely duplicated but really timing-tight identically
duplicated otherwise
it won't work. That would take 2x more LUTs and would not be practical for
maintenance - what would be changed on one module must be changed on other too.

It is so because real signal is not completely predictable, trigger
event may be missed
and armed capture time may take indefinitely until contition is satisfied

emard, 2019-07-08:
regarding the bug and latency
for now just remove process P_horizontal_scroll
and leave only inside part, instead of registered
it will become combinatorial logic

I can probably make better solution to use another
free registered process to make the arithmetic so that
I don't chain 2 arithmetic operations, for better fmax

emard, 2019-07-08:
I think i fixed this 48-pixel alias just by fixing the
capture1shot module

captured_addr must have no latency for calculating
BRAM address that reads the buffer for video display.
I wasn't aware how important is this latency

hdl4fpga, 2019-07-08:
Everything is too tight ... I'll still working on the capture system ... I'm trying to solve all the requirement.
> would not be practical for maintenance - what would be changed on one module must be changed on
> other too.
That's the worse thing.

emard, 2019-07-08:
HI

can you give me a try - i will make an attempt to separate
scopeio_capture1shot.vhd -- will contain triggers counters and storage
write logic
but not the storage itself
scopeio_storage.vhd -- will contain BRAM storage, video addressing and
and scrolling logic

hdl4fpga, 2019-07-08:
That is pretty much what is on scopeio_capture.vhd. But I still have some issues I'm trying to solve

emard, 2019-07-08:
True, with additional write bus logic
currently in the storage I probably I can reduce it
down tp 4 lines from 1-shot trigger module to the storage module:
reset_wr_addr, increment_wr_addr, trigger_t0_wr_addr, wr_enable

imporatant fact that makes it difficult is that
trigger must be "intelligent" and give signal
to start recording at precisely
predicted moment about half of the buffer length
BEFORE actual event of trigger T=0 happens.

emard, 2019-09-05:
HI

This bug is again there. After introduction of dual samplle storage mode
the code in scopeio_storage module became more complex for treating dual
sample modes, sufficiently unreadably even without solving the 48-pixel fake
vertical trace alias on the left.

I don't know what to do to fix this trace alias now, is it really only way to introduce a lot of delay lines
(extensive use of "align" module) or is there some more elegant solution for treating this trace
alias?

hdl4fpga, 2019-09-05:
Is it exactly the same issue ?

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hdl4fpga, 2019-09-05:
What are the files in which I can look for ? And have you any image or
something ?

emard, 2019-09-05:
It's exactly the same like before.
files related are

  ../../library/scope/scopeio_capture1shot.vhd \
  ../../library/scope/scopeio_storage.vhd \
  ../../library/scope/scopeio_tds_1shot.vhd \

It's because of 1-2 clock delay of signals in scopeio_storage
you will see a mess of the code for handling dual sampling case
I tried to code it as short as possible but still...

ULX3S bitstream is here (for PS/2 mouse on US2 port over OTG adapter)
https://github.com/emard/ulx3s-bin/tree/master/fpga/scope/scope-12f-ps2-bug

hdl4fpga, 2019-09-05:
I'll check to see if  I can find it

hdl4fpga, 2019-09-05:
video_frm should be delay two cycles if you want it to control video_dv..
scopeio_video waits two cycles till the very first moment it asks for the
sample by sending video_frm. This gives BRAM enough cycles to send the
samples along with a video_dv.
I think that is the problem.

hdl4fpga, 2019-09-05:
Sorry I forgot to send you the file and the line
https://github.com/emard/hdl4fpga/blob/f49dd751f1e953f8fcf979833e080a7be8b39698/library/scope/scopeio_tds_1shot.vhd#L197

emard, 2019-09-06:
Thanks I delayed video_frm 2 cycles and it helped for almost all cases.

It works for everything except downsampling=0 when
double samples are packed to make buffer appear as double size.
In storage module I have to use additional clock cycle to assemble
such sample.

For downsampling=0 best I can get is to delay 2 or 3 cycles.
delay 2: alias line appears on the left side on the grid.
delay 3: alias line appears on the right side of the grid.
I can't get completely rid of alias line.

hdl4fpga, 2019-09-06:
downsampling=0 is a special case. I use the change in video_frm from 0 to 1
to know that is the first sample and send the first sample twice. Then,  I
copy the sample into a register. The next sample is combination of the
saved sample with the next one and I save that in the register.

Here is the code

https://github.com/hdl4fpga/hdl4fpga/blob/c8f1686b688e267cf5cd54d8a0db2bbd873aa872/library/scope/scopeio_capture.vhd#L217

dv2 is the signal that is used to detect the first sample.

And here is how I get dv2

https://github.com/hdl4fpga/hdl4fpga/blob/c8f1686b688e267cf5cd54d8a0db2bbd873aa872/library/scope/scopeio_capture.vhd#L132

I hope this helps

regards

emard, 2019-09-11:
I tried to get rid of alias but I couldn't get anything working
it stays all the time only can choose side either left or right
depending do I delay it 2 or 3 clocks

I tried several simple ways to either duplicate the samples
or hide the trace in case of rising edge of video_frm detected by shift register

if rising_edge(video_clk) then
  video_frm_shift <= video_frm & video_frm_shift(video_frm_shift'high downto 1);

video_frm_first <= video_frm_shift(1) and not video_frm_shift(0);

I tired to use it as video enable
video_dv <= video_frm_shift(0) and trace_visible and not video_frm_first;

and to make duplicate samples (rd_data_00)
        for i in 0 to inputs-1 generate
          rd_data_00(i*(2*sample_bits) to (i+1)*(2*sample_bits)-1)
          <= rd_data_ds((2*i+1)*sample_bits to (2*i+2)*sample_bits-1)
           & rd_data_ds((2*i+1)*sample_bits to (2*i+2)*sample_bits-1);
          rd_data_0(i*(2*sample_bits) to (i+1)*(2*sample_bits)-1)
          <= pr_data_ds((2*i+1)*sample_bits to (2*i+2)*sample_bits-1)
           & rd_data_ds((2*i+0)*sample_bits to (2*i+1)*sample_bits-1);
          rd_data_1(i*(2*sample_bits) to (i+1)*(2*sample_bits)-1)
          <= rd_data_ds((2*i+0)*sample_bits to (2*i+1)*sample_bits-1)
           & rd_data_ds((2*i+1)*sample_bits to (2*i+2)*sample_bits-1);

and use the duplicated in case of the first
        captured_data <= rd_data_ds when downsampling = '1'
                    else rd_data_00 when first = '1'
                    else rd_data_0  when rd_addr(rd_addr'high) = '0'
                    else rd_data_1;

but nothing works, alias line appears in case of downsampling = 0

emard, 2019-09-11:
Hmm or wait maybe I got something to get rid of this allias
with a IMHO acceptable dowside of hiding first sample
which I prefer more than have complicated code that makes
creates double from single sample and has more delay lines

emard, 2019-09-11:
Additional drawback now are only few pixels on the right side of the
grid in the case horizontal scroll is moved past the sample data available

Traces should completely be hidden but few alias pixes remain on the
right side of the grid

Those pixels are not too annoying and I'd left the for the sake of
code simplicity
or unless you offer some simple low-LUT effective solution to get rid of them :)

closed 2019-09-11 by emard

hdl4fpga, 2019-09-11:
> Hmm or wait maybe I got something to get rid of this allias
> with a IMHO acceptable dowside of hiding first sample
> which I prefer more than have complicated code that makes
> creates double from single sample and has more delay lines
Ill try to see what can i do when I finish with the textbox. Its almost
ready.
you shoudn't add any delay line. hidding the first sample is an option

emard, 2019-09-11:
ok I'd like to test textbox too.

As I understood your message, storage module has to treat another special case
with 1st sample which it must be assembled differently. Already it
treats special
case of even and odd samples.

I opted to not treat additional special case and not to modify
scopeio_storage module
at all but to take the simpler way by not drawing 1st sample.

hdl4fpga, 2019-09-11:
You can try textbox, it is on work branch .... It's still work in progress