hdl4fpga/hdl4fpga #142: Working with 1024x600 small HDMI display

prasimix, 2019-07-12:
I'd like to use ULX3S with a small HMDI display that comes these days with 1024x600 resolution. It seems that 800x600 (4:3) with HDMI is fading out from the market or they are unnecessarily expensive (compared to 16:9).

hdl4fpga, 2019-07-12:
I don't have such display but I can add the video mode if you get one to try and test.

prasimix, 2019-07-12:
Nice, I'll try to get one and let you know when I'm ready for testing.

goran-mahovlic, 2019-07-12:
I have 5 inch that is 800X480 if that helps for testing ...

hdl4fpga, 2019-07-12:
Yes ... I'll add the modes ... We have to ask EMARD what frequencies are
available for videos ... I'll add a couple of layouts

Is  this https://www.adafruit.com/product/2406 the display ?

emard, 2019-07-12:
My default suggestion would be you just to go to
modeline calculator https://arachnoid.com/modelines/
calculate default for 1024x600 60 Hz and get entry like this
# 1024x600 @ 60.00 Hz (GTF) hsync: 37.32 kHz; pclk: 48.96 MHz
Modeline "1024x600_60.00" 48.96 1024 1064 1168 1312 600 601 604 622 -HSync +Vsync

We can add this entry straight away but it may not neccessary be best
for your display. Average FPGA can't make exactly 48.96 MHz but 50 MHz.
Because of ths and depending on display, you may need to manually experiment
with numbers 1312 and 622 until your display shows exactly 60 Hz and
no flicker or similar and then let us know what you got so we can add proper
entry

hdl4fpga, 2019-07-12:
Thanks for the syncs ... does top level ULX3S have 50 Mhz frequency to connect to the sync ?
just curious

goran-mahovlic, 2019-07-12:
[This is my 5" display connected to ULX3S](https://ibb.co/c6JPpPh)

hdl4fpga, 2019-07-12:
Two segments or three segments @prasimix ?

emard, 2019-07-12:
I can make easily clock generate 50/250 MHz at ULX3S from 25 MHz
onboard that itself is plain simple. I can test it on normal monitor if I have
luck it will show some picture and then I can adjust refresh rate so monitor
reports 60 Hz exactly. If not, then on real display it must be tried

hdl4fpga, 2019-07-12:
Ok ... perfect ... as you know it better than I ... thanks

emard, 2019-07-12:
For a small display looking at details may be visually
"too small" so 1 or 2 segments should be just fine while
technically 4 segments are possible on this display

hdl4fpga, 2019-07-12:
ok ... do you want to sketch it ?

prasimix, 2019-07-12:
> Two segments or four segments @prasimix ?

How about just one (like @emard proposed)?

emard, 2019-07-12:
Yes I will first try default some modeline with normal monitor
and see if it will show the picture.

hdl4fpga, 2019-07-12:
perfect ... the customer is always right 😂😂😂.
Pretty much configurable ... the hard work is to make the display to sync.
that is what @emard is doing

emard, 2019-07-12:
I can get something fast and send goran bitstream to try,
otherwise I will fix it over the weekend. I have some junk
code on my branch about usb so I will mix it up dirty just to
try display and later I can cleanup for proper merge

emard, 2019-07-12:
I generated 50 MHz clock and took default modeline
for 48.96 and got the picture with 61.2 refresh rate on normal monitor
I will first send this bitstream to Goran for testing on actual display
then depending on what happens, we can adjust details later

emard, 2019-07-12:
Yeah it's not so easy, now I'm trying on 12F board and
monitor shows the correct picture but with constantly
present small window in the center of the screen
 "signal is out of range" :)

emard, 2019-07-12:
On my git there is video mode setting for 1024x600 60Hz from 50 Hz pixel clock
and 1 segment. There is vscale numbering bug, as traces should be 0.0 but are labeled
-8.0

hdl4fpga, 2019-07-12:
I'll check

emard, 2019-07-13:
When we are at video modes can I ask what
is the difference between  Rat and Cat?
And should we have all Rats and no Cats :)

hdl4fpga, 2019-07-13:
I "R" is for reduce sync timings, the "at" is a "@",  I got from a vesa doc
a long time ago. I don't remember having used "C" instead of "R"

hdl4fpga, 2019-07-14:
> On my git there is video mode setting for 1024x600 60Hz from 50 Hz pixel clock
> and 1 segment. There is vscale numbering bug, as traces should be 0.0 but are labeled
> -8.0

I wasn't able to find the video. Is this the link https://github.com/emard/hdl4fpga-bugs in which I have to look for it ?

emard, 2019-07-14:
I didn't make the picture of this bug.
vertical scale is completely shifted by -8.0

so undriven traces should be at label 0.0
but becase of bug, are at -8.0 for 1024x600 layout.

for 800x480 there's no bug and trace is at 0.0
as it should normally be

hdl4fpga, 2019-07-14:
Ohhh sorry again ... my fault ... I misread your comment.

hdl4fpga, 2019-07-14:
I think the bug is fixed ... It was an overflow on vertical scale

emard, 2019-07-14:
yes its fixed, thanx!

emard, 2019-07-16:
you can close this issue, mode is added and bugs fixed

closed 2019-07-16 by hdl4fpga