hdl4fpga/hdl4fpga #143: capture1shot flicker

emard, 2019-07-12:
HI

tested latest changes, picture now flickers with capture1shot
it's quite annoying while before it was nearly perfect with any signal

there's something new, capture_dv = 1 , what this parameter
exactly do to the display and can I use it somehow for better
deflickering?

hdl4fpga, 2019-07-13:
I found a bug between the trigger system and the downsampler and fixed. I was supposed that nothing should happen. 

capture_dv tells the video system that the data capture is valid. if it is ' 0' then no sample is plot for that capture_addr. It is planned to be used to display the data captured while the captured buffer isn't still completed or users scroll to invalid time-axis values.

emard, 2019-07-13:
I can test this fix!

1shot trigger deflickering logic tries to always keep content of the
buffer in valid state so capture_dv = '1' is good value

Yes it's good idea to disable trace plot few pixels around the
point in time when buffer wraparound jumps from negative to
positive time, also the proper numbering of -min +max on
hzscale too

In capture1shot, the wraparound is always near capture_addr = 0
and T=0 in the middle of the buffer, at storage_length/2

emard, 2019-07-15:
I still need to keep 1shot deflicker=false (disabled) because if enabled it will licker
(it didn't flicker before some display changes).

A simplified logic is currently in use which will also deflicker
and It's normally useable but it can't honor "strobe" parameter
during triggering.

hdl4fpga, 2019-07-15:
Try the last fix. Even though I don't think it'll fix your problem ... I came to a conclusion the last week that it is impossible to fulfill all the user requests: on trigger systems, downsampler systems (that should be called interpolation/decimation) and capture system. So if the last fix doesn't solve the problem, it is better to duplicate the trigger and downsampler and set to the older version inside the generate ... and try to figure out and interface that let users choose among: to create theirs or to use the ones that fit better for them.

emard, 2019-07-15:
I will try the fix.

I agree that it is not possible to have all the features and keep the
code small and simple as we option "deflicker=false"
and signal is looking good with such setting.

So, this is not a serious issue - please keep the code small and
simple and don't duplicate anything.

hdl4fpga, 2019-07-15:
Yes ... that is my main goal

emard, 2019-07-15:
I found the thing which broke my deflicker algorithm.

I need regular timing from downsample signal as a regular
clock enable for deflickering because I must apply real time
calculation to external input signal.

What is currently done in downsample module seems to me
that downsample signal is mixed somehow in if/then loop with
trigger level shot signal, and after such mixing it's impossible to have
realtime deflickering calculation correct to stabilize signal in the buffer

For example, if input downsampler is doing 1:10 reduction,
then signal I need is 1-clock of '1' and 9-clocks of '0' regularly
all the time like this
100000000010000000001000000000....

hdl4fpga, 2019-07-15:
Can you check if it is fixed ?

emard, 2019-07-15:
checking...

emard, 2019-07-15:
no current patch makes everything work for some time
and then stop

In downsampler, the "downsample_dv" must not
depend on triggering - can you check is this code
in downsampler really correct:
                               if input_shot='1' and shot_dis='0' then

If instead of downsample_dv  I replace ampsample_dv
the 1shot capture module is able to stabilize the signal
but downsampling doesn't work then

hdl4fpga, 2019-07-16:
No .. it doesn't but output_sample is delayed 
can you try using resizedsample_data instead of  downsample_data ?

emard, 2019-07-16:
it didn't help.

for 1-shot trigger deflickering, it doesn't matter how much is the actual data
delayed because in storage module, I have option to compensate any latency 
with "align_to_grid" parameter

but, In downsampler scaler_p I deleted everything else except this:

if scaler(0)='1' then
 scaler := unsigned(factor);
else
 scaler := scaler - 1;
end if;
output_dv <= scaler(0);

then everything about deflickering started to work again
signal is stable in the buffer!

hdl4fpga, 2019-07-16:
ok ... save it ... I'll move downsampler into the generate block and will see later.

hdl4fpga, 2019-07-16:
Add the one that works for you into your generate block and do a pull request

emard, 2019-07-16:
I added linear deflickering mode which works and can
replace currently broken differential mode. I can later see
can I fix differential.

closed 2019-07-16 by emard