hdl4fpga/hdl4fpga #157: Multiple triggering inputs

emard, 2019-07-18:
HI

What is the intended principle of trigger with multiple inputs?

Should it track all channels and do some logical combination on
of each channel trigger_shot like AND/OR/XOR/NOT to make combinatorial trigger_shot
or should it do some channel math on sample values like CH1+CH2-CH3*CH4... and then
trigger or a user-defined function of everything mentioned :)?

Currently my GUI remembers trigger settings (level,edge) for
each channel and if I switch channel, also the trigger switches
to setting for appropriate channel.

hdl4fpga, 2019-07-18:
I came out with multiple trigger while I was thinking Fourier analysis. I don't have a clear idea. One solution is to save the values of the register trigger and  restore them without thinking what is inside of it as you are doing now.

emard, 2019-07-18:
then I can simplify GUI code when
now scopeio internal registers remember the
trigger settings for currently selected channel.

This simplification is however small because GUI
must still internally remember all trigger settings,
but now it doesn't need to re-send them after channel
change.

I was thinking as we have enough channels, it is
best that user in toplevel module creates user-defined
math function which combines sample values from
many real channel and creates a math channel, example
power channel current*voltage, or difference channel
voltage1-voltage2

such function will generate a synthetic channel which
can be triggered as any other real channel

closed 2019-07-18 by emard

hdl4fpga, 2019-07-18:
It might be I understood you wrong, but scopeio only save values for one channel only and that is the best way, I think, .
When I referred to multiple triggers, I thought of having two signal triggered by different signals at the same time. I wasn't referring to having different modes. For example you have an OFDM signal(wifi, dvb-t, etc) you want it to be trigger by a self correlation function, and you want to get the spectral analysis of that signal as a Fourier Transform FFT, The user will want to trigger the FFT by slope or whatever but it is trigger by itself. So you have two different control trigger signals at the same time

>I was thinking as we have enough channels, it is
>best that user in toplevel module creates user-defined
>math function which combines sample values from
>many real channel and creates a math channel, example
>power channel current*voltage, or difference channel
>voltage1-voltage2

This is why it is important to define carefully a scopeio_trigger entity that let user replace it with their own.

emard, 2019-07-18:
I see, you want to somehow trigger to an interesting part
of DVB-T by using certain self-correlate function which can
then generate triggering signal. I never did this, but probably
I would consider a solution in the toplevel as a module or
process for DVB-T input (I/Q probably) that will create a synthetic
triggering channel which I will use as trigger in the usual way then.

I was focusing my 1-shot trigger system to replicate
old-school behavior of analog scopes and this is intended
to stabilize periodic functions (sinewaves etc) on the screen
and to monitor and capture simple digital packet traffic like
RS232, PS/2 or USB which may look for e.g. CRC fail or NAK
as trigger so they must capture what happened before and after
the trigger condition.