hdl4fpga/hdl4fpga #159: XY mode

emard, 2019-07-18:
HI

Is XY mode possible (for lissajous, magnetic histeresys etc).
I'm not *really* sure if it can be done in any simple way

I was thinking about dual port BRAM as scan-line buffer.
Length of 2 video scan lines is buffered, one line is being output
for display while the other is being drawn with pixel content
traces grids etc.

Such concept will (I hope) completely get rid you from delay (align module)
because pixels can be written asynchronously from display during one scan line.

The BRAM port for line buffer when it displays every pixel it can can also clear
the pixel with background color at the same clock cycle on when this pixel is gone
to monitor for display

this method is called write-while-read (or destructive read, some memories in hardware
internally must actually destroy a bit in order to read it).

It use this for video 2D acceleration (sprites) and it works on BRAMs at spartan6, artix7,
cyclone4, ecp5, fx2, machox2

hdl4fpga, 2019-07-18:
The trace system only writes vertical lines and its very simple and can't do much more. I don't think It is able to do that.

For me, everything is about cost ... I don't know how much the cost is  in resources, I mean the BRAM and the control system compare to what it is working now. Graphics modules(grid, axis, traces, etc) have different latencies and align balance all the data to be just in time to be combined. If I understood it correctly, every module has to write its data to the memory at its time for the video to display then. But they will end only with one port of the BRAM as the other is accessed by the video system, so only one port will be shared by several modules which want to write and in the end the frequency of this modules that depend on the video  should be increased because the graphic modules generates the data with different latencies

I am not a lover of align ... but it is flexible enough ... it works as a FIFO ... any solution that gives flexibility and gets rid of it is welcome

emard, 2019-07-19:
well yes that is true, only one BRAM port has to be shared by different
modules and that can introduce a nasty multiplexer which will cut off fmax

closed 2019-07-29 by emard