hdl4fpga/hdl4fpga #165: Merging trigger downsampling and capturesystem

hdl4fpga, 2019-07-28:
@emard we have to merge those systems ....I think they are finish ... Tell me when you think I can pull a request from work repository.
I wrote s wiki about the new behavior here https://github.com/hdl4fpga/hdl4fpga/wiki/About-Trigger,-Down-sampling-and-Capture-system

regards

emard, 2019-07-29:
I will pull the latest stuff, hope it will at least compile :)

emard, 2019-07-29:
I have usbmouse update ready it works much better!
SIE module is in verilog from opencores.
I Deleted old unreliable usbmouse code.

emard, 2019-07-29:
Tomorrow I can pull work and try (gone sleeping now :)

emard, 2019-07-29:
HI

I can't compile work branch (unmerged)

@E: CD145 :"/tmp/hdl4fpga/library/common/align.vhd":39:2:39:3|Formal, "di", and actual agree on type but not on size
@E: CD145 :"/tmp/hdl4fpga/library/scope/scopeio.vhd":321:22:321:35|Formal, "di", and actual agree on type but not on size
@E: CD145 :"/tmp/hdl4fpga/library/common/align.vhd":40:2:40:3|Formal, "do", and actual agree on type but not on size
@E: CD145 :"/tmp/hdl4fpga/library/scope/scopeio.vhd":321:22:321:35|Formal, "do", and actual agree on type but not on size

hdl4fpga, 2019-07-29:
I've created a new branch ... I called it **merge**. I've left only the new system ... It seemed to compile ... Can you tried it and tell me if it works ? so then the functionality of the older one can be added

emard, 2019-07-29:
HI

Yes it compiles, and everything works, I have tested
it with oled and usb mouse, gui and video works.

in some h-zoom mode, new trace rendering system
seems to render traces differently, I "think" I have even
seen some trace rendering bug, but it's early to say is
it really so

Trigger is useable but it could be better :)

hdl4fpga, 2019-07-29:
ok ... ill merge to master and will be polishing bugs

thanks

emard, 2019-07-29:
Let it go, if I see the bugs again, I'll report!

hdl4fpga, 2019-07-29:
done!

emard, 2019-07-30:
bug: At 800x480 video, traces are drawn over hzscale, vtscale and textbox

hdl4fpga, 2019-07-30:
Besides that my monitor cannot deal well with the sync trimmings https://github.com/hdl4fpga/hdl4fpga/blob/master/doc/issues/images/480p.jpg. I cannot make it do it wrong. 
Those are the modes https://github.com/hdl4fpga/hdl4fpga/blob/f8deef12fe0ee1a9dca8d8b680fc95874a49d29f/library/scope/scopeiopkg.vhd#L145 and https://github.com/hdl4fpga/hdl4fpga/blob/f8deef12fe0ee1a9dca8d8b680fc95874a49d29f/library/scope/scopeiopkg.vhd#L241 , aren't they ?

Here is my test https://github.com/hdl4fpga/hdl4fpga/blob/2b74737ef164f6e868c4c397f3301c1d65e112b1/s3starter/scopeio/scopeio.vhd#L119. Have I done something wrong ? Xtal is 50 Mhz divided by 5 a multiplied by 3, 30 Mhz is gotten. But sync seems fine.

emard, 2019-07-30:
hmm I away now and here I have only this small screen
800x480 with me (30 MHz pixel clock)

I see the similar traces issue at OLED 96x64 maybe that could
you try?

emard, 2019-07-30:
I can't test it again but when I was last trying modes
even without much timing tweaking, my lenovo monitor
accepted 800x480 60Hz 30MHz

hdl4fpga, 2019-07-30:
I got this using 800x600 sync timmings:

1.  96x64

https://raw.githubusercontent.com/hdl4fpga/hdl4fpga/master/doc/issues/images/96x64at800x600.jpg

2. 800x480p

https://raw.githubusercontent.com/hdl4fpga/hdl4fpga/master/doc/issues/images/480pat800x600.jpg

emard, 2019-07-30:
HI

Hm on 96x64 I think I see trace going over vtscale
on 800x480 not sure.

Can I recommend to post large images to another git repository
hdl4fpga one is rapidly growing over 100 MB, I need
few backup copies and I'm slowly going out of disk space :)

Maybe not immediately, but in some future, a git cleanup purge
should be done to remove accumulated binaries...

hdl4fpga, 2019-07-30:
I had that behavior before I merged the new down sampling system. I've compared to master again and everything seems to be fine.

I cannot understand.

I will move the images.

hdl4fpga, 2019-07-30:
On that 96x64, the scale is inside the grid  and that is why traces are there.

emard, 2019-07-30:
reoving images are not enough to clalim space as git will
remeber them still  and just double the space using diff again
for removal)

a rebase with hard cleanup is needed (backup everything 2 times
before trying this)
https://help.github.com/en/articles/removing-sensitive-data-from-a-repository

emard, 2019-07-30:
To easier reproduce the bug I made 640x480 mode
which should be standard for any monitor 25 MHz pixel clock

I didn't make the timing as I don't have monitor here
but I guess you can load it to any higher mode to see the
traces bug.

emard, 2019-07-30:
HI

I'm trying to reconstruct special case when downsampling = '0'
can/are the data be so structured that without knowing the number
of input channels I can handle this special case e.g. with something
like this

        captured_data <= rd_data_ds when downsampling = '1'
                    else rd_data_ds(0 to rd_data_ds'length/2-1)
                       & rd_data_ds(0 to rd_data_ds'length/2-1)
                         when rd_addr(rd_addr'high) = '0'
                    else rd_data_ds(rd_data_ds'length/2 to rd_data_ds'high)
                       & rd_data_ds(rd_data_ds'length/2 to rd_data_ds'high);

hdl4fpga, 2019-07-30:
If I understood well, Have you taken into account that rd_addr should be delay 2 cylcles because of data delay ? That is the only thing I can think of now.
The other things seem to be right.

hdl4fpga, 2019-07-30:
Yep ... I've found something ... I'm working on it.

hdl4fpga, 2019-07-31:
Sorry ... It was the monitor not synchronizing correctly. ... Everything seems to be fine
https://github.com/hdl4fpga/hdl4fpga.github.io/raw/master/images/hdl4fpga/issues/vesa640x480.jpg

Those are Xilinx ISE ... I'll check on ecp3versa

hdl4fpga, 2019-07-31:
Definitely a Diamond Lattice bug ... it started to trash ... I'll go on tomorrow

emard, 2019-07-31:
aiieee diamond bugs again :(
well I'm glad that you can reproduce this!

I nearly wanted to discard 2x buffer but I supported it with some
bit shuffling... BRAM is small and it's good to have larger buffer sometimes :)

yeah delay, there is some minor bug with align to grid 1-2 pixels off depending
on downsampling=0/1, I can fix that later, it's easy

I have another bug at "click to trigger", I wonder how a simple GUI function
can hold so much bugs :(

hdl4fpga, 2019-07-31:
I think I found the bug 
https://github.com/hdl4fpga/hdl4fpga/blob/dfbfb3723124eda38b16a23c8c7a7265238d8d28/library/common/rom.vhd#L63. Xilinx wants it to be signal or crash, Lattice Diamond wants it to be constant or generates garbage. I set it constant to try.
This is ScopeIO running at 640x480 on ecp3versa https://github.com/hdl4fpga/hdl4fpga.github.io/raw/master/images/hdl4fpga/issues/ecp3versa640x480vesa.jpg

It seems everything to be right.

emard, 2019-07-31:
HI
I pulled the git but the bug is still the same

traces look similar to
hzaxis_height = vtaxis_width = 0
but the actual values are 8 and 6*8 :)
So actually traces should be hidden there but they are drawn

when we are at value 0, the hzscale is placed
on top of the grid - I can support that on the GUI
if neccessary but if it's not a big deal I'd like
if you can place hzaxis below the grid
on the same position when hzaxis_height = 0
as with hzaxis_height = 8

hdl4fpga, 2019-07-31:
> HI
> I pulled the git but the bug is still the same
>
> traces look similar to
> hzaxis_height = vtaxis_width = 0
> but the actual values are 8 and 6*8 :)
> So actually traces should be hidden there but they are drawn
>
> 😣  ... something is still broken ... I'll try to figure it out.

> when we are at value 0, the hzscale is placed
> on top of the grid - I can support that on the GUI
> if neccessary but if it's not a big deal I'd like
> if you can place hzaxis below the grid
> on the same position when hzaxis_height = 0
> as with hzaxis_height = 8

I'll try to do that after finding the bug. hzaxis is on top because it is
easier to compare to 0, but the cost to compare to anything else should be
the same.

hdl4fpga, 2019-08-01:
I couldn't find the bug that you told me ... but I found several weird thing tool bugs. I'am currently using Lattice Diamond 3.10.2.115 on ecp3versa. The bug I'm going to describe made bitstream not generating even the sync timings : 

if I instanced **scopeio** with 

`generic map (vlayout_id => 0)`

it works perfectly but if I declared a 

`constant layoutid : natural := 0;`

and then used it 

`generic map (vlayout_id => layouyid)`

I got a complete black screen 

So, I think **scopeio** is making the tool get its limits. I think there is a **buffer overflow or stack overflow somewhere on Lattice Diamond**.

emard, 2019-08-01:
your vhdl writing style is advanced, tools are surprized and unready :-)

I'm using diamond 3.7 and this is the last known to me
that synthesizes f32c up to 100 MHz
yes I also think there is a lot of memory leak and bad pointers in diamond.
from time to time, it leads to segmentation fault, for example
this completely crashed the diamond I made a typo:
      if R_rx_done <= '1' then
a hard crash without even a line number where it happend,
I fixed it with
        if R_rx_done = '1' then
compiled ok

You have ulx3s now so you can also try ecp5 maybe you find some
clue

emard, 2019-08-01:
Apart from that I'd suggest to also port to some altera cyclone4 or
whatever you have handy. every one of them Xilinx,Altera,Lattice will
dump errors at different parts of the code. In my experience, 2 of them
are not enough to tell is the code wrong or the compiler

hdl4fpga, 2019-08-01:
I have to try on it ... that's the next thing

emard, 2019-08-01:
here are some binaries to get started
also the scopeio bitstreams are there so you can try
https://github.com/emard/ulx3s-bin

hdl4fpga, 2019-08-01:
True

hdl4fpga, 2019-08-02:
I compiled https://github.com/hdl4fpga/hdl4fpga/blob/master/ULX3S/diamond/ulx3s_12f_scope.ldf ti check the bug and got this error.

52111106	ERROR - EHXPLLL 'clk_25M/clk_video_cpu_v_inst/pll_i' output CLKOP with -90 degree phase shift should not be used as the feedback signal (FEEDBK_PATH = INT_OP).

emard, 2019-08-02:
Let me see and fix

emard, 2019-08-02:
I generated this clock with my diamond 3.7
If it's still uncompileable, you may need to generate such clock with
your local diamond version....

hdl4fpga, 2019-08-02:
It compiled

emard, 2019-08-02:
I'm here to help

hdl4fpga, 2019-08-02:
It compiled and programmed ... ;) I'm sending you a picture in the next one

emard, 2019-08-02:
wow

hdl4fpga, 2019-08-02:
Is this the bug ? https://github.com/hdl4fpga/hdl4fpga.github.io/raw/master/images/hdl4fpga/issues/mybebug.jpg

hdl4fpga, 2019-08-02:
There is white line around

emard, 2019-08-02:
no but bug looks very similar to this,
you have to set
hzaxis_height = 8, vtaxis_width = 6*8

in my default source currently bug is hidden,
as I have set height = 0, vtaxis_width = 0

to see the bug, instead of
lcd800x480ongrid
you have to use
lcd800x480

however visually it will not look too much different anyway :)

hdl4fpga, 2019-08-02:
Now I see it

hdl4fpga, 2019-08-02:
if you set **trig1shot** to false, is the bug still there ?

emard, 2019-08-02:
yes

hdl4fpga, 2019-08-02:
This what I get when I set **trig1shot** to false, and  **constant vlayout_id: integer := 8;** that is **lcd800x480** ... I've  only change those constant.https://github.com/hdl4fpga/hdl4fpga.github.io/blob/master/images/hdl4fpga/issues/layout%3D8.jpg. Yellow line is limited to the grid.

emard, 2019-08-02:
> This is what I get when I set **trig1shot** to false, and  **constant

hmmmmmmmm let me try this
in this case trig1shot should be responsible :(

emard, 2019-08-02:
Yes with trig1shot disabled, traces are indeed limited to the
grid. Now I wonder what do I have missing in trig1shot
and what your trigger has to make it work correctlyy

emard, 2019-08-02:
there is this signal capture_av which I don't understand
what data should I output if its 0 or 1

emard, 2019-08-02:
I think I understand, this is trace enabler from video system
I need to somehow construct "invisible" data :)

emard, 2019-08-02:
can I have small help?
if I want to generate off-screen sample value for trace to be invisible
when capture_av=0, do I have some constant in some package
which in sample date makes this sample invisible?

hdl4fpga, 2019-08-02:
Of course ... that was I had meant with _"we have to merge"_

hdl4fpga, 2019-08-02:
And I was waiting for you to ask me about your trigger functionality to merge both

hdl4fpga, 2019-08-02:
And I'haven't realized that you had added yours again :))))))))) .... till yesterday

emard, 2019-08-02:
I overlooked the complexity and thought I will understand
everything easily just from the code,
but obviously it it's more complicated than
at the first glance, aiiieeeee :(

hdl4fpga, 2019-08-02:
First I want to make a merge of both functionalities to move on and then
I'll write a wiki to explain it ... I'll help me to understand what myself
from the past did ... :)))))))))

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

emard, 2019-08-02:
trig1shot itself works great, i am using it to support USB mouse
and it is able to catch all events the same as normal storage scope would

My storage module needs to generate "invisible"
samples so the traces don't appear at the hzscale I tried
few tricks but trace is always visible somewhere. I don't
understand how your "capture" module makes it invisible
out of grid.

For now, I can set all sample values to 0, this can be
a workaround.

this will mark the 0 position of each trace, so it can even have
a practical purpose, however being an ugly solution until
we have something better

emard, 2019-08-02:
dont merge this pull requesst i must clean

[120 more replies]

closed 2019-08-12 by hdl4fpga