hdl4fpga/hdl4fpga #183: hzaxis_height=vtaxis_width=0

emard, 2019-08-01:
on 800x480 I tried to make 3 segments of 32-pixel grid
but it can't fit because between segments some unused space
is allocated
                        num_of_segments  =>    3,
                        division_size    =>   32,
                        grid_width       =>   32*25, -- 800
                        grid_height      =>   32*5, -- 160 (*3 = 480)
                        axis_fontsize    =>    8,
                        hzaxis_height    =>    0,
                        vtaxis_width     =>    0,

emard, 2019-08-01:
ups forget there's gap 4 :)

emard, 2019-08-01:
The grid is drawn over the font, prehaps the font should have priority over the grid

hdl4fpga, 2019-08-01:
It is possible to do that ...Compact mode(when the scale is inside the grid) was described quick and dirty, but It resulted not to be so dirty, i.e ... low resources and it works pretty fine :)

emard, 2019-08-01:
I'm using it regularly and kinda liked it.
full screen utilization, no unused areas no borders :)

thanks to recent bug I finally support from mouse2rgtr
and now the bug is hidden :)

emard, 2019-08-03:
"bug" has gone, can the hzaxis numbers be moved
at the bottom of the grid, so that they are on the same
position (or similar) for both cases
hzaxis_height = 0, vtaxis_width = 0
and
hzaxis_height > 0, vtaxis_width > 0

hdl4fpga, 2019-10-04:
Moved ... and textbox can be put inside the grid by setting texbox_within to true.

closed 2019-10-04 by hdl4fpga