hdl4fpga/hdl4fpga #208: RMII 2-bit support

emard, 2019-08-27:
HI

Is udpipdaisy module supposed to accept 2-bit RMII data?

I connected LAN8720 RMII that has 2-bit TX/RX at 50 MHz
So i created
signal mii_rxdata,  mii_txdata  : std_logic_vector(0 to 1);

and tried to pass that to "scopeio_udpipdaisy" module but it
seems to refuse it, expecting 8 bits:
@E: CD297 :"/home/guest/src/fpga/hdl4fpga/library/scope/scopeio_udpipdaisy.vhd":94:53:94:60|Width mismatch, location has width 8, value 2

emard, 2019-08-27:
> Yea ... but all the data on the daisy chain should be 2 bits ... Have you
> tried to connect scopeio_miiudp directly to scopeio without anything else ?

Yes, similar problem:
@E: CD297 :"/home/guest/src/fpga/hdl4fpga/library/scope/scopeio_miiudp.vhd":116:5:116:11|Width
mismatch, location has width 8, value 2

emard, 2019-08-28:
> I don't know why compilers tells that the problem is there. It should
> complain on this line
>
> https://github.com/emard/hdl4fpga/blob/bb08adea44fded00cd6994004a8d73e2ac049dd0/library/scope/scopeio_udpipdaisy.vhd#L94
>
> I think the problem is here ... it should be 2 bits too.
>
> https://github.com/emard/hdl4fpga/blob/bb08adea44fded00cd6994004a8d73e2ac049dd0/ULX3S/scopeio/scopeio_top.vhd#L161

Hey thanx for the hint, now I realize that 2-bit vector
should be passed all the way thru :)!
Let me try

emard, 2019-08-28:
yeeesss I got IP address and ARP reponse using the 2-bit
vector all the way. Now I have some 8-bit issues with mouse2rgtr
but that's my code to fix

emard, 2019-08-28:
Maybe I should reopen :)
scopeio_miiudp accepts 2-bit RMII, obtains IP address and sends
ARP replies. But it seems to me that data output stream doesn't work 
so_dv and so_data displayed on OLED are all 0

emard, 2019-08-28:
I wanted to say that so_dv and so_data do change but resulting value
in shift register that accumulates them is 0

hdl4fpga, 2019-08-28:
It's strange .... I'll check

emard, 2019-08-28:
Yes if you can check this a bit.
What I'm sure that works - DHCP and ARP are doing well!

hdl4fpga, 2019-08-28:
Have you reverse  so_data ?

emard, 2019-08-28:
Could be, I must check,
I'm preparing to show them in HEX and scope traces to see them better.

What I'm also thinking,
as I have 2-bits coming at a time and in one point they have to be converted
to byte, then rxvalid signal is intermittent so I cannot have frm = rxvalid
to signalize one continuous packet, this was possible in 8-bit mode
but not 2-bit now

hdl4fpga, 2019-08-28:
Ohh .. you are converting it to 8 bits already ? ... Can you try first to
connect it directly to scopeio si_data with 2 bits to see if it works?

hdl4fpga, 2019-08-28:
Everything I've check is working ... Sadly I don't have a the a phy with 2
bits data to check, but I think is has something to do with bit order. On
phy bit 0 is the left most.

emard, 2019-08-28:
Yes first I'm trying to use them directly
no conversion at my side

I will try to send some network packet that changes
background color as a test.

At RMII side bit order is correct, if I swap them then DHCP is not working.
At output of upd module - maybe bit order should be swapped this I will check...

hdl4fpga, 2019-08-28:
Yep ... I was refering to at output of upd module

emard, 2019-08-28:
at module scopeio_miiudp
I am monitoring with scope lines
                so_dv       => net_fromistreamdaisy_irdy,
                so_data     => net_fromistreamdaisy_data

I see preamble and at least visually looking properly gated irdy
signal aligned to the packet payload. Maybe I have wrong clock
domain or something there's becoming real mess of trying to
support everything from ulx3s toplevel :)

emard, 2019-08-29:
Yeeeeesss output of scopeio_miiudp must be passed thru
reverse(data) and then it started to work :)

hdl4fpga, 2019-08-29:
WoW .... Nice! Good to know It's  working 😀 ... I haven't tried 2 bits yet

emard, 2019-08-29:
you can get It cheap, 6$ from ebay

It's a pity it doesn't have transformer, so
the module needs to be powered first
without ethernet cable. I have PoE switch
and must plug cable afterwards, other way
it won't work.

https://www.ebay.com/itm/LAN8720-Ethernet-Board-Network-Module-Ethernet-Transceiver-RMII-Interface/183833709567?_trkparms=aid%3D333200%26algo%3DCOMP.MBE%26ao%3D1%26asc%3D20171012094517%26meid%3D2015c25f83e243f9abe9c35cd4a6afa9%26pid%3D100008%26rk%3D1%26rkt%3D12%26sd%3D261479108838%26itm%3D183833709567%26pmt%3D1%26noa%3D0%26pg%3D2047675&_trksid=p2047675.c100008.m2219

emard, 2019-08-29:
I found it even cheaper for 2.2$ :)
https://www.ebay.com/itm/1pcs-Smart-Electronics-LAN8720-network-module-Ethernet-transceiver-for-arduino/183479331440?hash=item2ab8386a70:g:VgsAAOSwZPdbvcND

hdl4fpga, 2019-08-29:
Thanks.

Should I close this issue ?

emard, 2019-08-29:
Yes actually I can click close too :)

closed 2019-08-29 by emard