hdl4fpga/hdl4fpga #213: compiling upcoming textbox on lattice diamond

emard, 2019-09-13:
HI 

I got this error, don't know am I missing some package file to include
in the project or compiler compatibility issue

@E: CD255 :"/tmp/hdl4fpga/library/scope/scopeiopkg.vhd":554:20:554:21|No identifier "style" in scope
@E: CD255 :"/tmp/hdl4fpga/library/scope/scopeiopkg.vhd":1281:56:1281:57|No identifier "style" in scope

hdl4fpga, 2019-09-13:
I'll check

hdl4fpga, 2019-09-13:
Mine is working ... I compiled it using DIAMOND version 3.11.0.396.4. As I don't think we have the same version can you try qualifying the expression ... that is : to

(tagid_row, style => analogtime_rowstyle, ref => 0)

add

**tag'**(tagid_row, style => analogtime_rowstyle, ref => 0)

The **'** should be there between the **tag** and **(**

if that works I'll add it for compatibility

hdl4fpga, 2019-09-14:
I've done a pull request on your work repository if you want to try textbox.
It compiles syntax check but I get a run of memory on DIAMOND for ULX3S project.
It compiles well on ecp3versa project.

May I have an older code

emard, 2019-09-16:
HI

I will first try txtbox from your "work" code with suggested tag fix

emard, 2019-09-16:
Using your "work" repository, I just added tag'( and project files
that were missing but got vhdl internal error. Those are last messages printed:

@N: CD630 :"/tmp/hdl4fpga/library/scope/scopeio_storage.vhd":12:7:12:21|Synthesizing
hdl4fpga.scopeio_storage.beh.
@W: CD638 :"/tmp/hdl4fpga/library/scope/scopeio_storage.vhd":44:8:44:13|Signal
wr_ena is undriven. Either assign the signal a v
@N: CD630 :"/tmp/hdl4fpga/library/common/bram.vhd":32:7:32:10|Synthesizing
hdl4fpga.bram.inference.

While running 64-bit - Linux build
@E:Internal Error in c_vhdl

@E::Errors, while synthesis of top - ulx3s.
@END

emard, 2019-09-16:
is this module commited to "work" repository?
entity hdl4fpga.video_vga

emard, 2019-09-16:
Reason of vhdl error is initialization of shared variable,
with this it makes it compile again.
I can try to change init function to make it pass if possible

        shared variable ram : word_vector(0 to 2**addr_size-1); -- :=
init_ram(bitrom, 2**addr_size);

emard, 2019-09-16:
HI this is the workaround which works.
Apart from that, if retval is updated in for loop then
it compiles only this
 retval(i) := (others => '0');
and for example it doesn't compile
 retval(i) := (others => '1');

                aux(0 to bitrom'length-1) := bitrom;
                if bitrom'length > 1 then -- WORKAROUND
                for i in retval'range loop
                        retval(i) := aux(i*retval(0)'length to
(i+1)*retval(0)'length-1);
                end loop;
                end if; -- WORKAROUND
                return retval;

emard, 2019-09-16:
After above fix your "work" repository compiles and then
text box appears approx like this :)

*********    Trigger
********* *s *********
  Hz offset  :   Hz div
  Vt  offset  :   Vt div
********* *V *********
********* *V *********
********* *V *********
********* *V *********

hdl4fpga, 2019-09-16:
changing the values of vertical channels and horizontal values , it should
show them on the textbox

emard, 2019-09-16:
Vertical scale changes I see the trace waveform becoming
vertically larger or smaller but there are no numbers at hzscale/vtscale
and also textbox values don't change

hdl4fpga, 2019-09-16:
ok ... you said that thay is my work repository .... i made somthing
yestarday thay might not work

ill check

emard, 2019-09-16:
Yes it is your work repository recently synced.
You can update tag'( this compiles now and works
also this bram fix is ok to avoid initialization when
by default not required.

hdl4fpga, 2019-09-16:
I've made the fixes on work.

emard, 2019-09-21:
HI

I downloaded latest hdl4fpga "work" branch. It compiles
for ULX3S. Something is a bit changed in rgtr interface.
Horizontal zoom and scroll works. Vertical scroll works
but zoom behaves somehow unexpected, seems like
some rgtr bits changed meaning. Numbers on horizontal
and vertical scale are not visible (black on black?)

On textbox numbers don't change with changing horizontal zoom
(but waveform changes)

hdl4fpga, 2019-09-21:
> HI
>
> I downloaded latest hdl4fpga "work" branch. It compiles
> for ULX3S. Something is a bit changed in rgtr interface.
> Horizontal zoom and scroll works. Vertical scroll works
> but zoom behaves somehow unexpected, seems like
> some rgtr bits changed meaning.

I haven't tested with any lattice system yet

I've noticed it ... nothing should change ... It seems I broke something
when I was coding the textbox.
I am going to fix it when I finish textbox

Numbers on horizontal

and vertical scale are not visible (black on black?)

That's strange ....I changed palette subsystem to add user programmable
priority so axis letter can be over the grid.
But ti is supposed to be backward compatible

> On textbox numbers don't change with changing horizontal zoom

This is even more strange ... I've tested it on Xilinx with mouse an was
workind perfectly. Even the arrow changed when I changed the trigger edge
with mouse.

emard, 2019-09-22:
I think it is "normal" that when doing some larger project
code on one platform and compiling it on other platform it doesn't
work the same.

usually the portability changes are small but take effort to find.

In scopeio, most ofthen issue I remember is bit ordering,
is it 0 to N, N downto 0, reverse() or normal I never can guess
until I try :)

hdl4fpga, 2019-09-22:
Yep ... Ill take that into account when I start to port it to ULXS3.
I still have work to do. Now I solved most of the axis scale configuration.
I will focus on textbox again. I hope it is easy to make it work

hdl4fpga, 2019-10-01:
For Diamond Lattice, if I don't explicitly assign this signal, it is '0'.
https://github.com/hdl4fpga/hdl4fpga/blob/5902a3c5fb5b23933b94b59faa510600de583954/library/scope/scopeio_axis.vhd#L58
I've tested on ULX3S ... your turn ;-).

emard, 2019-10-01:
yes, some wanted default 1 or even random for
a more predictable code behavior, but lattice insists at 0 :)

big thanx for the ecp5 intervention, I'll download
and test new update with txtbox

emard, 2019-10-01:
HI

new txtbox looks absolutely beautiful congratulations!

I'm puzzled again how did you fit all of this functionality in
almost no more LUTs usage than before :)

That are good parts, well but mouse-rgtr interface got now bugs now
so I got to find what's up now. When I select anothey channel-trace
(which is not 1st)
then something strange happens with vertical settings. Otherwise when
1st channel
is selected, everything seems ok

Is anything important changed in rgtr binary format?

emard, 2019-10-01:
currently i "think" vertical bug is outside of my code (e.g mouse2rgtr etc)
but I allow possibility that this "theory" will be proven wrong wrong .... :)

here's how to reproduce and what happens then.
I show 2 traces 1=yellow and 2=cyan and control gui with mouse.

0. use hostmose.py to control the gui (hostmouse is not important,
same thing happens with ps2 or usb mouse)

1. drag yellow vertical scale numbered strip
(place cursor at vertical scale, hold left mouse button and move
vertical while holding)
It should normally work yellow trace should move
as you drag the vertical slider strip and txtbox will be updated.

2. place cursor and lower left yellow box in the segment and rotate
mouse wheel 1 step up.
Frame color should switch to cyan, indicating 2nd trace
is selected

3. drag cyan vertical scale - something strange will happen,
vertical cyan numbers will move as dragged but actually yellow
trace will move at half speed and in one point of dragging it will
disappear from screen
but will reappear if dragged back.

hdl4fpga, 2019-10-01:
> big thanx for the ecp5 intervention, I'll download
You're welcome

hdl4fpga, 2019-10-01:
> HI
>
> new txtbox looks absolutely beautiful congratulations!
Thanks ... It's still buggy .... It has some problems with colors and
something with the vertical scale.

>
> I'm puzzled again how did you fit all of this functionality in
> almost no more LUTs usage than before :)

I use a simple arbiter to usu the same simple float to bcd conversion.
So there is no need to replicate the same hardware unless that is what's
needed.

In fact that was the root of the problem ... bcdirdy - bcd initiator ready
- was '0', telling the unit to wait ... so nobody could use the unit ...
:-) .... It took me a couple of days ... It made crazy ...I didn't know
where to look for ...  I set it to '1' explicitly in a desperate act of "I
have to do something". Luckly I had the ULX3S, thank you very much for it
... that serial port saved me.

> That are good parts, well but mouse-rgtr interface got now bugs now
> so I got to find what's up now. When I select anothey channel-trace
> (which is not 1st)
> then something strange happens with [...]
I changed it a little bit to make TextBox work better and bugs came with
them ;-).
Working on bugs now that Textbox starts to work on ULX3S .

hdl4fpga, 2019-10-01:
> Thanks ... It will help to solve them.

> currently i "think" vertical bug is outside of my code (e.g mouse2rgtr etc)
> but I allow possibility that this "theory" will be proven wrong wrong ....
> :)
True ... working on them

emard, 2019-10-02:
Oho great that's already known bug being fixed
Yes I'm glad that ULX3S helped you in faster debugging,
this board is designed to be a small FPGA developer friendly,
with almost everything developer needs, full of useful low-cost features.
That's how I debug USB, controlling gui with mouse over serial port

There's also OLED onboard and my HEX decoder core so if you need to
monitor some integers, just connect data bus and hex will appear on screen,
similar to 7-segments but smaller and better :)....

emard, 2019-10-02:
A small hint with using hostmouse on linux (debian, gnome)

ctrl-alt-f3 switch from X11 gui to text console mode
(clicking on scopeio will not activate anything on linux desktop)

alt-f2 switch back from text console to X11 gui

hdl4fpga, 2019-10-02:
I think the bugs are fixed, many of them were due to my attempts to find the bug caused by  lattice diamond.
I can confirm that the my genius mouse needs 5V!. Xilinix's board has a  PS/2 connector with a JP that can configure mouse voltage supply to 3.5 V or 5 V. 3.5V can't make the mouse work.

I think most of the bugs are gone, if you can confirm that, I'll merge to master.

emard, 2019-10-03:
Let me test a little bit the absence of the bugs :))

Aha good experiment, this mouse is 5V-only
while ULX3S US2 port is 3.3V... I should have
included a compatible mouse in the first shipment :).

In the meantime I'm trying to rewrite USB PHY
New clock recovery method is much better, I'd say it doesn't loose
any bits at full speed (12 Mbit) but when I replace this new PHY,
something is different, it doesn't work ... needs more debugging

emard, 2019-10-03:
HI this is Logitech M-BT58 optical mouse on ebay for 3$
It works in both USB and PS/2 modes on ULX3S
Only drawback is that for PS/2 mode to work, it must be plugged
after bitstream is running

https://www.ebay.com/itm/Genuine-Logitech-M-BT58-Optical-Wheel-Mouse-USB-Wired-Black/183705799743?epid=1239246432&hash=item2ac5b80c3f:g:794AAOSw8IpcdhKo

emard, 2019-10-03:
HI

I have pulled your latest "work" branch and I can confirm
that bug is gone, both channels do scroll vertically and behave
normal as before

When merging I expect to appear a small mess because some
commits I did are just copied to "work" branch instead of being
pulled thru git somehow...

only important is that final code works and you can do whatever
you think is easy and good_enough method to resolve merging issues

I think better to waste time on core programming than git administration...

emard, 2019-10-03:
> One thing I'd like to see if >200 MB git repository could somehow
> be reduced to let's say 2-10 MB as there's I guess some binaries
> buried inside....
>
> du -h * | tail
> 520K

I tried merging and it worked out pretty clean :)
only diamond project was in conflict but this is
auto-generated by the scripts anyway.

After compiling it looks quite functional :)

hdl4fpga, 2019-10-03:
> One thing I'd like to see if >200 MB git repository could somehow
> be reduced to let's say 2-10 MB as there's I guess some binaries
> buried inside....
Most of them images ... I thought they weren't downloaded as they were
deleted .
You sent me some commands to clean them up but I lost them. Sure, bugs made
me be busy  😀

closed 2019-10-03 by hdl4fpga