hdl4fpga/hdl4fpga #219: textbox_width=0

emard, 2019-10-04:
if textbox is disabled by setting its width to 0,
ULX3S build fails with this error
@E: CD870 :"/home/guest/src/fpga/hdl4fpga/library/scope/scopeiopkg.vhd":1400:5:1400:16|Index into null range vector

hdl4fpga, 2019-10-04:
Yep .... check the **work** repository ... and set textbox_wthiin to true, I hope you like it . It's no t finished yet and as always I'm open to suggestions

hdl4fpga, 2019-10-04:
Fixed ... if user does't want it .. it will be turn off

closed 2019-10-04 by hdl4fpga

emard, 2019-10-05:
I compiled, "within" option of texbox, it looks really beautiful!

I need to adjust mouse2rgtr geometry constants
for hzaxis moved down - some pull request will follow :)

Maybe my additional "wish" would be to colorize
related lines in textbox for vertical channels to the
same color as trace colors...

hdl4fpga, 2019-10-05:
They are coming :))