hdl4fpga/hdl4fpga #222: display geometry strangeness

emard, 2019-10-06:
maybe it's not a bug but on "sd600" layout I changed this
and hzscale numbers have sunken down (only 1 top line
of the font shown - practically  unreadable) also text messages
moved towards center somehow.

Here's the bitstream for hostmouse.py
https://github.com/emard/hdl4fpga-bugs

-                       grid_width      => 16*32+1,
-                       grid_height     =>  8*32+1,
+                       grid_width      => 12*32+1,
+                       grid_height     =>  6*32+1,
-                       sgmnt_margin    => (top => 2, bottom => 2, others => 1),
-                       sgmnt_gap       => (horizontal => 1, others => 0)),
+                       sgmnt_margin    => (top => 4, bottom => 4, others => 1),
+                       sgmnt_gap       => (horizontal => 3, othe-                       sgmnt_margin    => (top => 2, bottom => 2, others => 1),
-                       sgmnt_gap       => (horizontal => 1, others => 0)),
+                       sgmnt_margin    => (top => 4, bottom => 4, others => 1),
+                       sgmnt_gap       => (horizontal => 3, others => 0)),
rs => 0)),

emard, 2019-10-06:
for oled 96x64 I can't compile project, setting
        constant vlayout_id: integer := 7;
leads to error

@E: CD751 :"/mt/scratch/tmp/fpga/hdl4fpga/library/scope/scopeiopkg.vhd":1415:5:1415:16|Index 2 is out of range: [1 to 1]

hdl4fpga, 2019-10-06:
I'll check it

hdl4fpga, 2019-10-07:
> maybe it's not a bug but on "sd600" layout I changed this
> and hzscale numbers have sunken down (only 1 top line
> of the font shown - practically unreadable) also text messages
> moved towards center [...]

This is fixed

hdl4fpga, 2019-10-07:
> for oled 96x64 I can't compile project, setting
> constant vlayout_id: integer := 7;
> leads to error
> 
> @e: CD751 :"/mt/scratch/tmp/fpga/hdl4fpga/library/scope/scopeiopkg.vhd":1415:5:1415:16|Index 2 is out of range: [1 to 1]
> 
> thing that fixes is setting
> textbox_width => 0,
> (it was 8 instead of 0)

It is fixed too ... I forgot to test it ... For a font with of 8 bits ... there is no enough room for textbox to fit in ... That cause the error ...

hdl4fpga, 2019-10-07:
Sorry ... I was wrong ... I set textbox_width with the wrong value ... anyway it's been fixed

emard, 2019-10-08:
Hmm I'd say bug is still around
text is positioned more towards the center of the grid with
current set of layout 1

     grid_width      => 12*32+1

but if I change

     grid_width => 24*32+1

Then text position is correct, aligned to the right

emard, 2019-10-08:
Can you also use "within=true" for vertical/horizontal scales placing
over the grid.

Maybe instead of "within" some more inuitive name like
hzscale_on_grid=true or hzscale_over_grid=true

I prefer the width's to have actual value instead of 0 so I can use
more readable math for width in mouse2rgtr geometry calculation
to determine clickable areas.

emard, 2019-10-08:
this makes yellow-filled segments, no grid no traces :)

                sd600 => (
                        display_width   =>  800,
                        display_height  =>  600,
                        num_of_segments =>    2,
                        division_size   =>   32,
                        grid_width      => 24*32+1,
                        grid_height     =>  7*32+1,
                        axis_fontsize   =>    8,
                        hzaxis_height   =>  0, --  8,
                        vtaxis_width    =>  0, --1*8,
                        vttick_rotate   => ccw90,
                        textbox_width   => 32*8,
                        textbox_within  => true,
                        main_margin     => (left => 0, top => 23, others => 0),
                        main_gap        => (vertical => 16, others => 0),
                        sgmnt_margin    => (top => 4, bottom => 4, others => 5),
                        sgmnt_gap       => (horizontal => 0, others => 0)),

hdl4fpga, 2019-10-08:
It's a feature not a bug :)) ... to make it use fewer resources as
possible, the code rounds to closest power of two slice to fit the textbox.
Also if you use a textbox width that is not a power of two you'll see a
strange behavior too. I'll code it  in a way that if the parameters meet
the constraints, it'll select one or the other. But now I'm focusing on
colors :)).

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#m_1077730026992256311_DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

hdl4fpga, 2019-10-08:
numbers are over the grid but background is bellow the grid.

emard, 2019-10-08:
Ahhaaa it's a code saving feature this power of 2 -> I agree to keep
the code short.
Also the naming within = over(numbers) &/ under(background) seems logical :)

great then lets do colors :)

emard, 2019-10-08:
OK

but I currently have difficulty trying to make arithmetic +/- combination
of geometry constants to calculate clickable rectangular areas for mouse.
Whatever I try when I fit on one combination then when I change constants
ffor margins and gaps, it doesn't fit anymore...

hdl4fpga, 2019-10-09:
Bugs fixed and suggestions done

closed 2019-10-09 by hdl4fpga

emard, 2019-10-10:
HI but iz grid_x calculated properly in case of vtaxis_within = true ?
someho I think vtaxis_width(layout) should not be added for within = true hmm

                retval := retval + vtaxis_x(layout);
                retval := retval + vtaxis_width(layout);
                retval := retval + layout.sgmnt_gap(horizontal);

hdl4fpga, 2019-10-10:
i forgot that ....

hdl4fpga, 2019-10-10:
i tested and everything was working ... have you tried the last master ?

hdl4fpga, 2019-10-10:
i'll fix that ... sorry i got confused
thanks

hdl4fpga, 2019-10-10:
I think I fixed them.