hdl4fpga/hdl4fpga #227: Lattice Diamond for Windows crashes when opening ULX3S project

gojimmypi, 2019-10-11:
When opening `hdl4fpga\ULX3S\diamond\ulx3s_12f_scope.ldf` in Lattice Diamond 3.10.3.144 for Windows, Diamond quietly crashes. I'm awaiting Diamond 3.11 download to see if that will help,

The problem seems to be with `\hdl4fpga\library\video\cgafonts.vhd`; renaming this file with a `.bad`  extension and Diamond no longer crashes (although other errors, probably related to missing file - like the undeclared `psf1cp850x8x16`).

hdl4fpga, 2019-10-12:
Yeah ... It seems It has to do with Windows Lattice GUI. It does not crash on any Linux Lattice GUI. It seems that they are still using some sort of 16 bits Windows 3.1 library when files larger than 64Kb were prohibited. Lattice Diamond bug. WIndows Xilinx ISE and Windows Vivado can handle the file with no problem. I am still not able to solve the problem. That file has the vhdl description of the fonts.

gojimmypi, 2019-10-12:
Indeed you are correct. Reducing the file size to about 2300 lines and Diamond does not crash.

my VHDL skills are rather weak. Do you think something like this would work:

```
package cgafonts is
	constant psf1cp850x8x16 : std_logic_vector(0 to 32767) := (
	  psf1cp850x8x16part1 +
	  psf1cp850x8x16part2
	  );
end;
```

with `psf1cp850x8x16part1` and `psf1cp850x8x16part2` defined in separate files?

hdl4fpga, 2019-10-12:
Good idea!  ... don't use parenthesis and concatenate with *&*

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gojimmypi, 2019-10-12:
ah! yes, append with &

like this?

```
package cgafonts is
	constant psf1cp850x8x16 : std_logic_vector(0 to 32767) := 
	  psf1cp850x8x16part1 &
	  psf1cp850x8x16part2;
end;
```
Stay tuned for PR.

hdl4fpga, 2019-10-12:
That should do it

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gojimmypi, 2019-10-12:
ok, I need some assistance here, please.

I've split the `cgafonts` into [separate files](https://github.com/gojimmypi/hdl4fpga/tree/master/library/video/cgafonts), but I am unable to integrate the VHDL changes into the main project. I ended up manually editing the `ulx3s_12f_scope.ldf` [here](https://github.com/gojimmypi/hdl4fpga/blob/master/ULX3S/diamond/ulx3s_12f_scope.ldf#L327) as the windows version of the IDE kept wanting to do weird things. One that I cannot seem to prevent, is the `Implementation` section changing `def_top`  [here](https://github.com/gojimmypi/hdl4fpga/blob/master/ULX3S/diamond/ulx3s_12f_scope.ldf#L5) from:

```
  <Implementation title="project" dir="project" description="project" synthesis="synplify" default_strategy="Strategy1">
    <Options def_top="ulx3s" top="ulx3s"/>
```

to

```
    <Implementation title="project" dir="project" description="project" synthesis="synplify" default_strategy="Strategy1">
        <Options def_top="hdl4fpga.ulpi_wrapper_vhdl" top="ulx3s"/>
```
I do not know if this change is related to some of the problems described below.

Here are the errors I see

```
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(76,71-76,102) (VHDL-1120) indexed name is not a unsigned
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(117,45-117,60) (VHDL-1120) indexed name is not a std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(38,1-119,9) (VHDL-1284) unit rtl ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/library/video/cga_adapter.vhd(37,37-37,51) (VHDL-1241) psf1cp850x8x16 is not declared
	2049990	ERROR - C:/workspace/hdl4fpga/library/video/cga_adapter.vhd(34,1-53,5) (VHDL-1284) unit cga_adapter ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_axis.vhd(531,38-531,50) (VHDL-1241) psf1digit8x8 is not declared
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_axis.vhd(531,52-531,62) (VHDL-1241) psf1bcd4x4 is not declared
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_axis.vhd(77,1-570,5) (VHDL-1284) unit def ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(489,19-489,29) (VHDL-1272) type error near r_sawtooth ; expected type std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(535,23-535,24) (VHDL-1278) type unsigned does not match with the integer literal
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(662,40-662,48) (VHDL-1033) cannot read from 'out' object adc_mosi ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(664,39-664,46) (VHDL-1033) cannot read from 'out' object adc_csn ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(665,40-665,48) (VHDL-1033) cannot read from 'out' object adc_sclk ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(16,1-1417,5) (VHDL-1284) unit beh ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/library/uart/uart_rx.vhd(37,3-37,11) (VHDL-1081) formal clk_rate has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/uart/uart_rx_f32c.vhd(10,2-10,15) (VHDL-1081) formal c_clk_freq_hz has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/usb/usbcdc/usb_control.vhd(104,9-104,15) (VHDL-1081) formal nendpt has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_istreamdaisy.vhd(30,1-51,5) (VHDL-1423) design unit scopeio_istreamdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_udpipdaisy.vhd(31,1-59,5) (VHDL-1423) design unit scopeio_udpipdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_ps2mouse2daisy.vhd(10,1-36,5) (VHDL-1423) design unit scopeio_ps2mouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_usbmouse2daisy.vhd(11,1-50,5) (VHDL-1423) design unit scopeio_usbmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_hostmouse2daisy.vhd(11,1-31,5) (VHDL-1423) design unit scopeio_hostmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/video/cga_rom.vhd(8,1-19,5) (VHDL-1423) design unit cga_rom contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio.vhd(32,1-90,5) (VHDL-1423) design unit scopeio contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_capture.vhd(32,1-48,5) (VHDL-1423) design unit scopeio_capture contains unconstrained port(s)
```

I'm not sure if the other errors are related, but the one I caused is the `psf1cp850x8x16 is not declared`. I think I am not properly referencing the libraries and packages. Any tips on how to fix this?

The GitHub [diff](https://github.com/gojimmypi/hdl4fpga/commit/69f9daffd2968926dea0c3118bc3a646cd7a2243#diff-4933f14d78153261a755577db434d356R3) makes it appear there was a massive delete / replace to the ldf file... but really it looks more like this:
[image]

hdl4fpga, 2019-10-12:
I don't know how to fix that ... but I'll split cgafont.vhd myself in week ..

gojimmypi, 2019-10-12:
well, I think it is just because of my lack of experience with VHDL. I'm essentially trying to do a C#-like "partial public class".  If you can help me with a basic template of a given package file referencing constant declarations in two other files... I'd be happy to help here. 

I believe I have `cgafont.vhd` _split_ - it just needs to be _referenced_ properly. ;)

hdl4fpga, 2019-10-12:
The problem is the project file, that has been modified is broken, and it
has nothing to do with vhdl ... I don't know how to help you with that.
I rather like to have a complete solution ... I was avoiding to split that
file ... but sooner or later I will have to do it. Now I'm adding color to
the readling texbox and I think in week I can make cgafont.vhd work on
windows lattice diamond.

sorry

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

gojimmypi, 2019-10-12:
yes, you are correct: there was some sort of problem with the project file. I believe I have resolved that.

The splitting I did was to simply put the different declarations into [separate files in a new sub-directory](https://github.com/gojimmypi/hdl4fpga/tree/master/library/video/cgafonts).  In addition to addressing the problem of the file being too big, I think it makes it easier to understand. I'm certainly open to suggestions.

I believe I've addressed all of the `cgafont`-related errors, but I still see these other errors:

```
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(76,71-76,102) (VHDL-1120) indexed name is not a unsigned
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(117,45-117,60) (VHDL-1120) indexed name is not a std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/common/hdl/adc/max1112x_reader.vhd(38,1-119,9) (VHDL-1284) unit rtl ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(489,19-489,29) (VHDL-1272) type error near r_sawtooth ; expected type std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(535,23-535,24) (VHDL-1278) type unsigned does not match with the integer literal
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(662,40-662,48) (VHDL-1033) cannot read from 'out' object adc_mosi ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(664,39-664,46) (VHDL-1033) cannot read from 'out' object adc_csn ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(665,40-665,48) (VHDL-1033) cannot read from 'out' object adc_sclk ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga/ULX3S/scopeio/scopeio_top.vhd(16,1-1417,5) (VHDL-1284) unit beh ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga/library/uart/uart_rx.vhd(37,3-37,11) (VHDL-1081) formal clk_rate has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/uart/uart_rx_f32c.vhd(10,2-10,15) (VHDL-1081) formal c_clk_freq_hz has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/usb/usbcdc/usb_control.vhd(104,9-104,15) (VHDL-1081) formal nendpt has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_istreamdaisy.vhd(30,1-51,5) (VHDL-1423) design unit scopeio_istreamdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_udpipdaisy.vhd(31,1-59,5) (VHDL-1423) design unit scopeio_udpipdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_ps2mouse2daisy.vhd(10,1-36,5) (VHDL-1423) design unit scopeio_ps2mouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_usbmouse2daisy.vhd(11,1-50,5) (VHDL-1423) design unit scopeio_usbmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_hostmouse2daisy.vhd(11,1-31,5) (VHDL-1423) design unit scopeio_hostmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio.vhd(32,1-90,5) (VHDL-1423) design unit scopeio contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga/library/scope/scopeio_capture.vhd(32,1-48,5) (VHDL-1423) design unit scopeio_capture contains unconstrained port(s)
```
Note this is all off your `master` branch. Perhaps it would be better if I instead make changes from your `work` branch?

Let me know if you'd like me submit a PR to contribute what I have so far - and if so, to which branch.

hdl4fpga, 2019-10-12:
I've just compiled master branch and It works . I don't know what that could be ... I'll split the files and try to fix it for tomorrow ... if  you cannot make it work ... you can try it tomorrow.

Work branch may not work ...I use it as mobile storage

gojimmypi, 2019-10-13:
despite the errors it synthesizes for me as well.

[image]

gojimmypi, 2019-10-13:
It works!
[image]

hdl4fpga, 2019-10-13:
😄

hdl4fpga, 2019-10-13:
Fixed ... now Windows Lattice Diamond does not crash   ... cgafont.vhd was split into four part.  related to code pages.

Thanks!

closed 2019-10-13 by hdl4fpga

gojimmypi, 2019-10-13:
I fetched & pull these latest changes; Diamond for Windows does not crash, but the project does not synthesize, either.

```
log file: "C:\workspace\hdl4fpga-master\ULX3S\diamond\project\project_project.srr"
Running: hdl_info_gen in foreground

Generating HDL info...

Job: "hdl_info_gen" terminated with error status: 2
See log file: "C:\workspace\hdl4fpga-master\ULX3S\diamond\project\syntax.log"
# Sun Oct 13 05:45:25 2019

Return Code: 2
Run Time:00h:00m:01s
Error: At line 180 while processing "project_project_synplify.tcl"
2
TCL script complete: "project_project_synplify.tcl"
TCL script had errors: "project_project_synplify.tcl"
exit status=9
exit status=9
Save changes for project:
C:\workspace\hdl4fpga-master\ULX3S\diamond\project\proj_1.prj
batch mode default:no

Synthesis exit by 9.
Done: error code 9
```

Additionally, there are over 160 errors;
```
Done: design load finished with (165) errors, and (27) warnings
```
my PR (which does have nearly 20 errors), does however, work as noted above.

hdl4fpga, 2019-10-13:
Strange ... that means that Diamond stopped due to errors and I don't have
any. I've just compiled it again on a new fresh master's download and it
compiles smooth on both GUI: on Windows and  Linux.
Have you clean your process up ? It is under process menu on GUI.

gojimmypi, 2019-10-13:
Thanks for the tip: Indeed there must have been some sort of undesired leftovers. A fresh master's download *does* synthesize into a bit file, but curiously there are still errors in the `Refresh Design` output: 

```
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/common/hdl/adc/max1112x_reader.vhd(76,71-76,102) (VHDL-1120) indexed name is not a unsigned
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/common/hdl/adc/max1112x_reader.vhd(117,45-117,60) (VHDL-1120) indexed name is not a std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/common/hdl/adc/max1112x_reader.vhd(38,1-119,9) (VHDL-1284) unit rtl ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(489,19-489,29) (VHDL-1272) type error near r_sawtooth ; expected type std_logic_vector
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(535,23-535,24) (VHDL-1278) type unsigned does not match with the integer literal
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(662,40-662,48) (VHDL-1033) cannot read from 'out' object adc_mosi ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(664,39-664,46) (VHDL-1033) cannot read from 'out' object adc_csn ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(665,40-665,48) (VHDL-1033) cannot read from 'out' object adc_sclk ; use 'buffer' or 'inout'
	2049990	ERROR - C:/workspace/hdl4fpga2/ULX3S/scopeio/scopeio_top.vhd(16,1-1417,5) (VHDL-1284) unit beh ignored due to previous errors
	2049990	ERROR - C:/workspace/hdl4fpga2/library/uart/uart_rx.vhd(37,3-37,11) (VHDL-1081) formal clk_rate has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga2/library/uart/uart_rx_f32c.vhd(10,2-10,15) (VHDL-1081) formal c_clk_freq_hz has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga2/library/usb/usbcdc/usb_control.vhd(104,9-104,15) (VHDL-1081) formal nendpt has no actual or default value
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_istreamdaisy.vhd(30,1-51,5) (VHDL-1423) design unit scopeio_istreamdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_udpipdaisy.vhd(31,1-59,5) (VHDL-1423) design unit scopeio_udpipdaisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_ps2mouse2daisy.vhd(10,1-36,5) (VHDL-1423) design unit scopeio_ps2mouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_usbmouse2daisy.vhd(11,1-50,5) (VHDL-1423) design unit scopeio_usbmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_hostmouse2daisy.vhd(11,1-31,5) (VHDL-1423) design unit scopeio_hostmouse2daisy contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio.vhd(32,1-90,5) (VHDL-1423) design unit scopeio contains unconstrained port(s)
	2049990	ERROR - C:/workspace/hdl4fpga2/library/scope/scopeio_capture.vhd(32,1-48,5) (VHDL-1423) design unit scopeio_capture contains unconstrained port(s)

``` 

And thanks a lot for getting these changes in place! btw - do you have any Getting Started docs? There's mention that ScopeIO is controlled via UDP packets, but  by ULX3S does not have a network interface (erm, other than the one on the ESP32).

hdl4fpga, 2019-10-13:
That is something that Lattice Diamond always does when loading a project.
Nothing to worry about ... Tool analyses  each entity by itself without
taking into account that they are part of a whole system.

That's @EMARD 's work ... he got a ethernet-phy and connected to it by the
available pins on the board. Also he did a USB-ethernet PHY core for ULX3s
that cheated the PC to make it believe that it had connected an USB to
ethernet adapter. It is very interesting work that let users do research on
USB-ethernet adapters All those cores are maintained by him and were
uploaded on ULX3S directory

gojimmypi, 2019-10-20:
when you get a chance, could you please apply the split CGA fonts to the `work` branch? thanks.

hdl4fpga, 2019-10-20:
Done ... but work is not working now

hdl4fpga, 2019-10-20:
Sorry .. I meant at the moment