hdl4fpga/hdl4fpga #233: Frozen display vertical scaling

emard, 2019-10-17:
This is maybe a wanted or intuitively expected feature
maybe it's not so easy to implement

I gave bitstream to the standard user to test it and his
first objection was that during frozen display, vertical scale
numbers do change but traces don't resize.

How difficult is to implement it?
My workaround could be at GUI level to disable changing
of vertical scale when display is frozen

hdl4fpga, 2019-10-17:
An unexpected bug ... It's easy ... I'll implement it  after I finish the layout for readings. It's puzzle me more than I thought. 

Thanks

emard, 2019-10-17:
OK I will need to again fix mouse2rgtr,
it should all combination of hz/vt within/without

hdl4fpga, 2019-11-04:
Now scales don't change when freeze is selected.
Only a small bug on the sign on textbox horizontal unit ..  a negative appears.
i'll try solve it tomorrow

emard, 2019-11-04:
I have seen trigger negatives - a friendly bug, not worth being reported :)

hdl4fpga, 2019-11-04:
I think everything is fixed ...I have my highest hope :)))

emard, 2019-11-04:
Yes trace vertical scaling is now prevented when in frozen mode!
when I roll mouse wheel it will postpone scaling and jump to the
new scale when unfrozen. I wonder should also horizontal scaling
be prevented in the similar way as vertical

closed 2019-11-04 by emard

hdl4fpga, 2019-11-04:
Doesn't horizontal scale get frozen ? I think I' ve fixed it too
Is it broken now ?

emard, 2019-11-04:
Horizontal is also locked - you fixed both, I forgot to test :)

hdl4fpga, 2019-11-04:
You scared me :))))) .... I thought there were more bugs. you know it
wasn't easy to solve the it.

emard, 2019-11-04:
not scary, halloween is over :)
still bugs will have some chance to show
at implementation of HEX decoder :)