hdl4fpga/hdl4fpga #235: wishlist for txtbox

emard, 2019-10-19:
show mouse pointer X/Y position in terms of time/voltage (useful for freq/amplitude measurement)
rgtr command to show/hide txtbox (useful when txtbox is within the grid)

hdl4fpga, 2019-10-19:
Textbox aim is to add display values easier than printf :-)).

hdl4fpga, 2019-10-19:
Here, https://github.com/hdl4fpga/hdl4fpga/blob/3f71968a0e35a2d805cc787d919a674cb45b7da6/library/scope/textboxpgk.vhd#L452 there is an example of what is coming

hdl4fpga, 2019-11-02:
**The genie thought for a second, then asked, "Do you want that highway with two lanes or four?"**
Help yourself ... it was a long fight against synthesis tools ... especially Xilinx ... that is the worse but I made it work. You only have to tell me what are registers to display out but I think you can guess how it works by reading the code
https://github.com/hdl4fpga/hdl4fpga/blob/e8d04239ef26089960cb1edeefa1b551e97ad20d/library/scope/scopeiopkg.vhd#L585

hdl4fpga, 2019-11-02:
I needed to do that ... I had been working on it before you asked it .. if i had not  done that textbox  won't scale up

emard, 2019-11-02:
HI

I guess I first need to try the new stuff and then I can
ask some "good" questions, if I checkout "work"
branch and compile, will then the traces vertically scale even if
the trigger is in frozen state?

hdl4fpga, 2019-11-02:
I haven't done it yet ...
But now I have that that is finished for a while at least ... I'll fix it
those these days.
May be tomorrow.

emard, 2019-11-02:
Its great that a milestone is finished and working!

hdl4fpga, 2019-11-03:
thanks

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2019-11-03 by emard