hdl4fpga/hdl4fpga #240: wish txtbox hex decoder

emard, 2019-11-03:
HI

This is just a SF idea, maybe over-complicated...

I'd like to have simple protocol analyzer, text over the waveform
of usb or ethernet packets to display BIN or HEX byte content :)

Let's say user should obtain or write some protocol decoder core
like MII or ULPI, which outputs bytes, scopeio stores decoded values to BRAM.
Then scopeio txtbox engine at each triggering point transition displays such
bram content as HEX or BIN digits in place in time when they are decoded.

hdl4fpga, 2019-11-03:
Coming there ....

emard, 2019-11-03:
Yes that would be awesome!
To have a waveform _-_-_-__---____-----_
and below see 55 55 55 65 C0 A8 00 01

What I currently do to monitor UTMI is
connect 8-bit rxdata to yellow and 1-bit rxvalid to cyan trace
makes kinda "analog" view of rxdata. If there was a hex decoder
then when e.g. cyan is selected as trigger, at each crossing edge of
the triggering cyan line it would display decoded HEX
of yellow trace :)

hdl4fpga, 2022-04-30:
Done that

hdl4fpga, 2022-04-30:
Done that

closed 2022-04-30 by hdl4fpga