hdl4fpga/hdl4fpga #260: won't compile on diamond 3.7

emard, 2020-01-03:
HI

Many good wishes for New year currently running :)

I merged the recent changes but got this:

I see in the function everything related to error line is declared as constant so it's for
me not obvious what should be fixed then..:

@E: CD640 :"/mt/scratch/tmp/fpga/hdl4fpga/library/scope/scopeio.vhd":195:31:195:45|Real value must evaluate to a constant for synthesis

hdl4fpga, 2020-01-03:
Hello

That constant is used to set the vertical gains automatically. My diamond version is 3.11.* (free download)  and it works with no complain. I set the **vt_steps** in the **hdl4fpga.work/ULX3S/scopeio/scopio_top.vhd** level file which then set the **k** constant. if you are using for some reason other **scopeio_top.vhd** than the one that is on the **hdl4fpga** you have to set it, as **vt_steps** is set to a **null vector range value** by default. It is that way in case of full custom gain values.

Each value of **vt_steps** corresponds to the gain of each of  the input channel, **arty kit** does that mess. And it is the value that comes on the ADC datasheet Vref/2**N in which N is the number of ADC bits.  **vt_unit** is the grid's division for the maximum amplifier value, that is, the value that is set to see the least significant bits of the ADC. You can use pico, nano, micro or milli multiplied by a constant to set the desired value.

The **k** constant is the result of that calculation. It also checks that there is no overflow in the gain constants. 

Happy New year and nice your wishes are coming true :)

emard, 2020-01-03:
Thanx!

I have slightly customized toplevel, I will check your version toplevel first

hdl4fpga, 2020-01-03:
I modified **hdl4fpga.work/ULX3S/scopeio/scopio_top.vhd** to show how those values should be set.
I've noticed how busy you are with the new ULX3S applications.
Of course I expect you are going to send yours in the next pull request.

Regards

emard, 2020-01-03:
yes actually I wanted functional scopeio to debug my UDP->ESP32->PS/2 converter.

for retro computer emulations, we may need keyboard mouse and joystick, too much input
hardware to be copy-pasted on the desktop, so my plan is to use something like "hostmouse.py"
on linux to send user input events over the network to ESP32 and then it will emulate PS/2 and
similar input device protocols

hdl4fpga, 2020-01-03:
On Fri, Jan 3, 2020 at 2:37 PM emard <notifications@github.com> wrote:

> yes actually I wanted functional scopeio to debug my UDP->ESP32->PS/2
> converter.
>
NICE!!! .... I hope it really helps to make that work.

emard, 2020-01-04:
It helps a lot!

with scopeio it is very comfortable, I use my 1-shot trigger to
freeze the waveform and immediately I see the waveforms
can can adjust signal timings to look identical as the real PS/2 device!

closed 2020-01-06 by hdl4fpga