hdl4fpga/hdl4fpga #354: values on txtbox 2x smaller than on hzscale/vtscale

emard, 2020-06-10:
with division_size=16 (layout lcd480x272seg1)
offset values shown on the right of the screen in "txtbox"
have 2x less value than values printed on hzscale/vtscale
while they should be the same.

(offset values dynamically change as traces are scrolled
horizontally or vertically)

hdl4fpga, 2020-06-11:
Can you send me a picture by gitter ?

emard, 2020-06-11:
I'll try later! Here are steps how to reproduce:

set grid size to 16 and recompile

1. locate hzscale slider at the bottom, its values are reading e.g.
0.0 130.0 260.0 ...
2. locate textbox, first value below IP address reading e.g.
0.0: 32.5 us
3. with the mouse place cursor on hzscale number "1" of 130.0 and left-drag hzscale slider to the left so that 130.0 comes to the left of the screen where 0.0 was before.
4. observe textbox value now! It should display a value around 130 but it displays half of this value, in my case 66:
66.0: 32.5 us

Similar thing happens to vtscale too

If grid size is not 16 but 32, then the reading are correct so I'd
say that grid size is not taken into account in the txtbox formula.

hdl4fpga, 2020-06-15:
Fixed it

closed 2020-06-15 by hdl4fpga